cartographer-project/cartographer #33: only support LIDAR & IMU sensor input

amiltonwong, 2016-10-06:
Hi, authors,

Currently, the package seems only support LIDAR & IMU sensor input, will it be possible to be adaptive to camera input? such as RGB-D, mono, stereo?

Thanks~

damonkohler, 2016-10-06:
The cartographer_turtlebot repo has configurations for depth camera input. Currently, Cartographer only supports scan matching and therefore requires some sort of depth data. However, the scan matcher could be replaced/supplemented with something vision based.

I'll change the description of this issue to better reflect requested feature.

BrannonKing, 2017-09-22:
It's quite common for the depth cameras to support a point cloud output. However, the idea of supporting a mono (er, regular) camera is interesting.