cartographer-project/cartographer #36: Structs and classes containing eigen typed data members missing EIGEN_MAKE_ALIGNED_OPERATOR_NEW

catskul, 2016-10-06:
Per docs classes and structs containing eigen types require EIGEN_MAKE_ALIGNED_OPERATOR_NEW to be safely allocated on the heap.

https://eigen.tuxfamily.org/dox-devel/group__TopicStructHavingEigenMembers.html

The following incomplete list may violate this requirement:

```
$ find ./ -name "*.h*" | xargs grep --color "^[^(]*Eigen::[^(&={]*$"
./cartographer/transform/rigid_transform.h:  using Affine = Eigen::Transform<FloatType, 2, Eigen::Affine>;
./cartographer/transform/rigid_transform.h:  using Vector = Eigen::Matrix<FloatType, 2, 1>;
./cartographer/transform/rigid_transform.h:  using Rotation2D = Eigen::Rotation2D<FloatType>;
./cartographer/transform/rigid_transform.h:  using Affine = Eigen::Transform<FloatType, 3, Eigen::Affine>;
./cartographer/transform/rigid_transform.h:  using Vector = Eigen::Matrix<FloatType, 3, 1>;
./cartographer/transform/rigid_transform.h:  using Quaternion = Eigen::Quaternion<FloatType>;
./cartographer/transform/rigid_transform.h:  using AngleAxis = Eigen::AngleAxis<FloatType>;
./cartographer/mapping_2d/map_limits.h:  Eigen::Vector2d max_;
./cartographer/mapping_2d/xy_index.h:  Eigen::Array2i min_xy_index_;
./cartographer/mapping_2d/xy_index.h:  Eigen::Array2i max_xy_index_;
./cartographer/mapping_2d/xy_index.h:  Eigen::Array2i xy_index_;
./cartographer/mapping_2d/scan_matching/correlative_scan_matcher.h:typedef std::vector<Eigen::Array2i> DiscreteScan;
./cartographer/mapping_2d/scan_matching/fast_correlative_scan_matcher.h:  const Eigen::Array2i offset_;
./cartographer/mapping_2d/scan_matching/occupied_space_cost_functor.h:    Eigen::Matrix<T, 3, 3> transform;
./cartographer/common/math.h:  Eigen::SelfAdjointEigenSolver<Eigen::Matrix<double, N, N>>
./cartographer/mapping_3d/rotation_cost_function.h:  const Eigen::Quaterniond delta_rotation_imu_frame_;
./cartographer/mapping_3d/acceleration_cost_function.h:  const Eigen::Vector3d delta_velocity_imu_frame_;
./cartographer/mapping_3d/hybrid_grid.h:  const Eigen::Vector3f origin_;
./cartographer/mapping_3d/imu_integration.h:  Eigen::Vector3d linear_acceleration;
./cartographer/mapping_3d/imu_integration.h:  Eigen::Vector3d angular_velocity;
./cartographer/mapping_3d/imu_integration.h:  Eigen::Matrix<T, 3, 1> delta_velocity;
./cartographer/mapping_3d/imu_integration.h:  Eigen::Quaternion<T> delta_rotation;
./cartographer/mapping_3d/submaps.h:                        Eigen::Array2i* min_index, Eigen::Array2i* max_index);
./cartographer/mapping_3d/submaps.h:  std::vector<Eigen::Array4i> voxel_indices_and_probabilities_;
./cartographer/mapping_3d/scan_matching/fast_correlative_scan_matcher.h:  std::vector<std::vector<Eigen::Array3i>> cell_indices_per_depth;
./cartographer/mapping_3d/scan_matching/fast_correlative_scan_matcher.h:  Eigen::Array3i offset;
./cartographer/mapping_3d/scan_matching/rotational_scan_matcher.h:  Eigen::VectorXf histogram_;
./cartographer/mapping/sparse_pose_graph.h:      Eigen::Matrix<double, 3, 3> sqrt_Lambda_ij;
./cartographer/mapping/sparse_pose_graph.h:      Eigen::Matrix<double, 6, 6> sqrt_Lambda_ij;
./cartographer/mapping/submaps.h:  Eigen::Vector3f origin;
./cartographer/kalman_filter/unscented_kalman_filter.h:  Eigen::SelfAdjointEigenSolver<Eigen::Matrix<FloatType, N, N>>
./cartographer/kalman_filter/unscented_kalman_filter.h:  using StateType = Eigen::Matrix<FloatType, N, 1>;
./cartographer/kalman_filter/unscented_kalman_filter.h:  using StateCovarianceType = Eigen::Matrix<FloatType, N, N>;
./cartographer/kalman_filter/unscented_kalman_filter.h:    std::vector<Eigen::Matrix<FloatType, K, 1>> Z;
./cartographer/kalman_filter/pose_tracker.h:typedef Eigen::Matrix3d Pose2DCovariance;
./cartographer/kalman_filter/pose_tracker.h:typedef Eigen::Matrix<double, 6, 6> PoseCovariance;
./cartographer/kalman_filter/pose_tracker.h:  Eigen::Quaterniond orientation_;
./cartographer/kalman_filter/pose_tracker.h:  Eigen::Vector3d gravity_direction_;
./cartographer/kalman_filter/pose_tracker.h:  Eigen::Vector3d imu_angular_velocity_;
./cartographer/kalman_filter/pose_tracker.h:  using StateCovariance = Eigen::Matrix<double, kDimension, kDimension>;
./cartographer/kalman_filter/gaussian_distribution.h:  Eigen::Matrix<T, N, 1> mean_;
./cartographer/kalman_filter/gaussian_distribution.h:  Eigen::Matrix<T, N, N> covariance_;
./cartographer/sensor/compressed_point_cloud.h:  Eigen::Vector3f current_point_;
./cartographer/sensor/compressed_point_cloud.h:  Eigen::Vector3i current_block_coordinates_;
./cartographer/sensor/laser.h:  Eigen::Vector2f origin;
./cartographer/sensor/laser.h:  Eigen::Vector3f origin;
./cartographer/sensor/laser.h:  Eigen::Vector3f origin;
./cartographer/sensor/point_cloud.h:typedef std::vector<Eigen::Vector3f> PointCloud;
./cartographer/sensor/point_cloud.h:typedef std::vector<Eigen::Vector2f> PointCloud2D;
```

SirVer, 2016-10-07:
That is a great change, thanks for doing this. Did you see crashes when running the code without?

catskul, 2016-10-07:
Have not had a chance to look for crashes in Cartographer, but have seen crashes in other projects using Eigen that appear to have been alignment related.

Note though @wohe's comment here may apply:
https://github.com/googlecartographer/cartographer/issues/34