cartographer-project/cartographer #42: can not build the cartographer in x86 ubuntu 14.04

gongyue666, 2016-10-11:
error at:
[245/327] Building CXX object cartogra...dir/fast_correlative_scan_matcher.cc.o
FAILED: /usr/bin/c++    -O3 -DNDEBUG -isystem /usr/include/eigen3 -isystem /usr/local/include -I. -I.. -I../cartographer/common/{Boost_INCLUDE_DIRS} -isystem /usr/include/lua5.2     -O3 -DNDEBUG -std=c++11 -Wall -Wpedantic -Werror=format-security -Werror=return-type -Werror=uninitialized -MMD -MT cartographer/mapping_2d/scan_matching/CMakeFiles/mapping_2d_scan_matching_fast_correlative_scan_matcher.dir/fast_correlative_scan_matcher.cc.o -MF "cartographer/mapping_2d/scan_matching/CMakeFiles/mapping_2d_scan_matching_fast_correlative_scan_matcher.dir/fast_correlative_scan_matcher.cc.o.d" -o cartographer/mapping_2d/scan_matching/CMakeFiles/mapping_2d_scan_matching_fast_correlative_scan_matcher.dir/fast_correlative_scan_matcher.cc.o -c ../cartographer/mapping_2d/scan_matching/fast_correlative_scan_matcher.cc
In file included from /opt/eigen/Eigen/Geometry:39:0,
                 from /opt/eigen/Eigen/Dense:6,
                 from /usr/local/include/ceres/internal/numeric_diff.h:40,
                 from /usr/local/include/ceres/dynamic_numeric_diff_cost_function.h:70,
                 from /usr/local/include/ceres/ceres.h:44,
                 from ../cartographer/common/math.h:25,
                 from ../cartographer/mapping_2d/probability_grid.h:28,
                 from ../cartographer/mapping_2d/scan_matching/fast_correlative_scan_matcher.h:33,
                 from ../cartographer/mapping_2d/scan_matching/fast_correlative_scan_matcher.cc:17:
/opt/eigen/Eigen/src/Geometry/Rotation2D.h: In instantiation of ‘Eigen::Rotation2D<Scalar> Eigen::Rotation2D<Scalar>::operator*(const Eigen::Rotation2D<Scalar>&) const [with _Scalar = double]’:
../cartographer/mapping_2d/scan_matching/fast_correlative_scan_matcher.cc:270:73:   required from here
/opt/eigen/Eigen/src/Geometry/Rotation2D.h:78:28: error: could not convert ‘(((Eigen::Rotation2D<double>::Scalar)((const Eigen::Rotation2D<double>*)this)->Eigen::Rotation2D<double>::m_angle) + ((Eigen::Rotation2D<double>::Scalar)other.Eigen::Rotation2D<double>::m_angle))’ from ‘Eigen::Rotation2D<double>::Scalar {aka double}’ to ‘Eigen::Rotation2D<double>’
   { return m_angle + other.m_angle; }
                            ^
[245/327] Building CXX object cartogra...eal_time_correlative_scan_matcher.cc.o
FAILED: /usr/bin/c++    -O3 -DNDEBUG -isystem /usr/include/eigen3 -isystem /usr/local/include -I. -I.. -isystem /usr/include/lua5.2 -I../cartographer/common/{Boost_INCLUDE_DIRS}     -O3 -DNDEBUG -std=c++11 -Wall -Wpedantic -Werror=format-security -Werror=return-type -Werror=uninitialized -MMD -MT cartographer/mapping_2d/scan_matching/CMakeFiles/mapping_2d_scan_matching_real_time_correlative_scan_matcher.dir/real_time_correlative_scan_matcher.cc.o -MF "cartographer/mapping_2d/scan_matching/CMakeFiles/mapping_2d_scan_matching_real_time_correlative_scan_matcher.dir/real_time_correlative_scan_matcher.cc.o.d" -o cartographer/mapping_2d/scan_matching/CMakeFiles/mapping_2d_scan_matching_real_time_correlative_scan_matcher.dir/real_time_correlative_scan_matcher.cc.o -c ../cartographer/mapping_2d/scan_matching/real_time_correlative_scan_matcher.cc
In file included from /opt/eigen/Eigen/Geometry:39:0,
                 from /opt/eigen/Eigen/Dense:6,
                 from /usr/local/include/ceres/internal/numeric_diff.h:40,
                 from /usr/local/include/ceres/dynamic_numeric_diff_cost_function.h:70,
                 from /usr/local/include/ceres/ceres.h:44,
                 from ../cartographer/common/math.h:25,
                 from ../cartographer/mapping_2d/probability_grid.h:28,
                 from ../cartographer/mapping_2d/scan_matching/real_time_correlative_scan_matcher.h:44,
                 from ../cartographer/mapping_2d/scan_matching/real_time_correlative_scan_matcher.cc:17:
/opt/eigen/Eigen/src/Geometry/Rotation2D.h: In instantiation of ‘Eigen::Rotation2D<Scalar> Eigen::Rotation2D<Scalar>::operator*(const Eigen::Rotation2D<Scalar>&) const [with _Scalar = double]’:
../cartographer/mapping_2d/scan_matching/real_time_correlative_scan_matcher.cc:123:71:   required from here
/opt/eigen/Eigen/src/Geometry/Rotation2D.h:78:28: error: could not convert ‘(((Eigen::Rotation2D<double>::Scalar)((const Eigen::Rotation2D<double>*)this)->Eigen::Rotation2D<double>::m_angle) + ((Eigen::Rotation2D<double>::Scalar)other.Eigen::Rotation2D<double>::m_angle))’ from ‘Eigen::Rotation2D<double>::Scalar {aka double}’ to ‘Eigen::Rotation2D<double>’
   { return m_angle + other.m_angle; }
                            ^
[245/327] Building CXX object cartogra....dir/trajectory_connectivity_test.cc.o
ninja: build stopped: subcommand failed.
=======================
plz help~

damonkohler, 2016-10-11:
Could you tell us what version of Eigen you have installed?

gongyue666, 2016-10-11:
thanks for reply~the vision is 3.2.0-8 and I find some locks on the folders in Eigen, I don't know is normal or not?

gongyue666, 2016-10-11:
In another computer which is X64 ,but hardwares and other conditions are the same, I have succeeded to build and run the demo.I do not want to replace the computer because it is to boring to dismantle the robot.

damonkohler, 2016-10-11:
* Cartographer only support 64-bit systems. That doesn't explain the error message you saw though.
* The error message looks like you may be using a different compiler. Compare your results to the following:

```
$ /usr/bin/c++ -v
Using built-in specs.
COLLECT_GCC=/usr/bin/c++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.8/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.4-2ubuntu1~14.04.3' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3) 
```

gongyue666, 2016-10-11:
ok, good news is I can use cartographer though~
the result in my computer is : 
exbot@exbot:~$ /usr/bin/c++ -v
Using built-in specs.
COLLECT_GCC=/usr/bin/c++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/i686-linux-gnu/4.8/lto-wrapper
Target: i686-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.4-2ubuntu1~14.04.1' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-i386/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-i386 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-i386 --with-arch-directory=i386 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-targets=all --enable-multiarch --disable-werror --with-arch-32=i686 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=i686-linux-gnu --host=i686-linux-gnu --target=i686-linux-gnu
Thread model: posix
gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.1)

damonkohler, 2016-10-11:
That looks correct to me. I'm not sure what's causing this particular issue, but it's not worth pursuing anyway since the system is only 32-bit. If you have any issues on your 64-bit system, please open a separate issue. I'll close this n ow.

closed 2016-10-11 by damonkohler

alvarezmdavid, 2016-11-16:
Hi! i already compiled the package few days ago, it compiled well but now i'm trying to compile it again and i have an error about Eigen3

alvarezmdavid, 2016-11-16:
Hi! i already compiled the package few days ago, it compiled well but now i'm trying to compile it again and i have an error about Eigen3

ubuntutecnalia@UbuntuTecnalia-2016:~/cart_catkin_ws$ catkin_make_isolated --install --use-ninja
Base path: /home/ubuntutecnalia/cart_catkin_ws
Source space: /home/ubuntutecnalia/cart_catkin_ws/src
Build space: /home/ubuntutecnalia/cart_catkin_ws/build_isolated
Devel space: /home/ubuntutecnalia/cart_catkin_ws/devel_isolated
Install space: /home/ubuntutecnalia/cart_catkin_ws/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 5 packages in topological order:
~~  - ceres_solver (plain cmake)
~~  - cartographer (plain cmake)
~~  - cartographer_ros_msgs
~~  - cartographer_ros
~~  - cartographer_rviz
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

==> Processing plain cmake package: 'ceres_solver'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/ubuntutecnalia/cart_catkin_ws/build_isolated/ceres_solver/install'
ninja: no work to do.
==> ninja -j8 -l8 in '/home/ubuntutecnalia/cart_catkin_ws/build_isolated/ceres_solver/install'
[1/5] Performing update step for 'ceres_src'
==> ninja install in '/home/ubuntutecnalia/cart_catkin_ws/build_isolated/ceres_solver/install'
[2/2] Install the project...
-- Install configuration: ""
<== Finished processing package [1 of 5]: 'ceres_solver'

==> Processing plain cmake package: 'cartographer'
==> Building with env: '/home/ubuntutecnalia/cart_catkin_ws/install_isolated/env.sh'
==> cmake /home/ubuntutecnalia/cart_catkin_ws/src/cartographer -DCMAKE_INSTALL_PREFIX=/home/ubuntutecnalia/cart_catkin_ws/install_isolated -G Ninja in '/home/ubuntutecnalia/cart_catkin_ws/build_isolated/cartographer/install'
-- Build type: Release
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   iostreams
-- Found required Ceres dependency: Eigen version 3.2.0 in /usr/include/eigen3
-- Found required Ceres dependency: Glog in /usr/include
-- Found Ceres version: 1.11.0 installed in: /home/ubuntutecnalia/cart_catkin_ws/install_isolated
CMake Error at CMakeLists.txt:32 (find_package):
  By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen3", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/ubuntutecnalia/cart_catkin_ws/build_isolated/cartographer/install/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntutecnalia/cart_catkin_ws/build_isolated/cartographer/install/CMakeFiles/CMakeError.log".
<== Failed to process package 'cartographer': 
  Command '['/home/ubuntutecnalia/cart_catkin_ws/install_isolated/env.sh', 'cmake', '/home/ubuntutecnalia/cart_catkin_ws/src/cartographer', '-DCMAKE_INSTALL_PREFIX=/home/ubuntutecnalia/cart_catkin_ws/install_isolated', '-G', 'Ninja']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/ubuntutecnalia/cart_catkin_ws/build_isolated/cartographer && /home/ubuntutecnalia/cart_catkin_ws/install_isolated/env.sh cmake /home/ubuntutecnalia/cart_catkin_ws/src/cartographer -DCMAKE_INSTALL_PREFIX=/home/ubuntutecnalia/cart_catkin_ws/install_isolated -G Ninja

Command failed, exiting.

Help, thanks

alvarezmdavid, 2016-11-17:
Hi! i excecuted the ckeck command to see where i have eigen conections and the result was this:

ii  libarpack2                                            3.1.5-2                                             amd64        Fortran77 subroutines to solve large scale eigenvalue problems
ii  libeigen3-dev                                         3.2.0-8                                             all          lightweight C++ template library for linear algebra
ii  ros-indigo-eigen-conversions                          1.11.8-0trusty-20160627-190934-0700                 amd64        Conversion functions between: - Eigen and KDL - Eigen and geometry_msgs.
ii  ros-indigo-eigen-stl-containers                       0.1.4-0trusty-20160321-180829-0700                  amd64        This package provides a set of typedef's that allow using Eigen datatypes in STL containers
ii  ros-indigo-tf-conversions                             1.11.8-0trusty-20160628-094836-0700                 amd64        This package contains a set of conversion functions to convert common tf datatypes (point, vector, pose, etc) into semantically identical datatypes used by other libraries. The conversion functions make it easier for users of the transform library (tf) to work with the datatype of their choice. Currently this package has support for the Kinematics and Dynamics Library (KDL) and the Eigen matrix library. This package is stable, and will get integrated into tf in the next major release cycle (see roadmap).
ii  ros-indigo-tf2-eigen                                  0.5.13-0trusty-20160627-192052-0700                 amd64        tf2_eigen

i also executed: sudo apt-get install libeigen3-dev

ubuntutecnalia@UbuntuTecnalia-2016:~$ sudo apt-get install libeigen3-dev
[sudo] password for ubuntutecnalia: 
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
libeigen3-dev ya está en su versión más reciente.
0 actualizados, 0 se instalarán, 0 para eliminar y 2 no actualizados.

(the result was that 2 files where not upgraded)

then upgrade it:

ubuntutecnalia@UbuntuTecnalia-2016:~$ sudo apt-get upgrade libeigen3-dev
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
Calculando la actualización... Listo
libeigen3-dev ya está en su versión más reciente.
Se actualizarán los siguientes paquetes:
  libcgmanager0 yelp-xsl
2 actualizados, 0 se instalarán, 0 para eliminar y 0 no actualizados.
Necesito descargar 260 kB de archivos.
Se utilizarán 8.192 B de espacio de disco adicional después de esta operación.
¿Desea continuar? [S/n] S
AVISO: ¡No se han podido autenticar los siguientes paquetes!
  libcgmanager0 yelp-xsl
¿Instalar estos paquetes sin verificación? [s/N] S
Des:1 http://es.archive.ubuntu.com/ubuntu/ trusty-backports/main libcgmanager0 amd64 0.39-2ubuntu2~ubuntu14.04.1 [34,2 kB]
Des:2 http://es.archive.ubuntu.com/ubuntu/ trusty-backports/main yelp-xsl all 3.12.0-1~ubuntu14.04.1 [225 kB]
Descargados 260 kB en 0seg. (597 kB/s)
(Leyendo la base de datos ... 280313 ficheros o directorios instalados actualmente.)
Preparando para desempaquetar .../libcgmanager0_0.39-2ubuntu2~ubuntu14.04.1_amd64.deb ...
Desempaquetando libcgmanager0:amd64 (0.39-2ubuntu2~ubuntu14.04.1) sobre (0.24-0ubuntu7.5) ...
Preparando para desempaquetar .../yelp-xsl_3.12.0-1~ubuntu14.04.1_all.deb ...
Desempaquetando yelp-xsl (3.12.0-1~ubuntu14.04.1) sobre (3.10.1-1) ...
Configurando libcgmanager0:amd64 (0.39-2ubuntu2~ubuntu14.04.1) ...
Configurando yelp-xsl (3.12.0-1~ubuntu14.04.1) ...
Procesando disparadores para libc-bin (2.19-0ubuntu6.9) ...

it finished ok! but when i try to compile and i get the same error that i posted yesterday.

Thanks for the quick answer, hope you can help me to fix this, i'll keep trying to fix it by myself but i can use some help.

Sorry some commands are in spanish, thats my home lenguage.

Regards.
David

SirVer, 2016-11-17:
I actually saw this error today for the first time. It happened when somebody tried to install a full ROS installation in a ROS workspace. Are you using a system wide installed ROS?

Try this: Make sure you did not source any ROS `setup.bash`, i.e. `echo $ROS_DISTRO` should not show anything. Then follow http://wiki.ros.org/kinetic/Installation/Ubuntu, especially install `sudo apt-get install ros-kinetic-desktop-full`. After that `source /opt/ros/kinetic/setup.bash` and try building our stuff again. You have to start from scratch there - i.e. make a new ROS workspace.

alvarezmdavid, 2016-11-17:
I have three different workspaces, one of them have ethzasl_icp_mapping, velodyne, loam_velodyne velodyne_height_map on it, i builded this one with just catkin_make

The second one have the blam files (Berkeley SLAM), also builded with catkin_make

The third and last one just have cartographer and cartographer_ros, builded with catkin_make_isolated like the docs said it should be.

I have some command lines added by me in the /.bashrc, to source the first and the second workspace, might this be the problem?

> source /opt/ros/indigo/setup.bash
> 
> source ~/catkin_ws/devel/setup.bash 
> 
> source ~/blam_ws/devel/setup.bash 
>

 

Maybe the last two lines are the problem, right now im outside of the office and i cant make some tests, but i remember that executing echo $ROS_DISTRO showed me "indigo".

Maybe erasing the lines i added my problem will be solved?.

Thanks!

David.

alvarezmdavid, 2016-11-18:
Hello Holger.

I erase the command lines at mi .bashrc and unsourced all the catkin workspaces and follow the steps you adviced me to follow and the mission was failed. 

```
ubuntutecnalia@UbuntuTecnalia-2016:~$ echo $ROS_DISTRO

ubuntutecnalia@UbuntuTecnalia-2016:~$ cd cart_catkin_ws/
ubuntutecnalia@UbuntuTecnalia-2016:~/cart_catkin_ws$ catkin_make_isolated --install --use-ninja
catkin_make_isolated: no se encontró la orden
ubuntutecnalia@UbuntuTecnalia-2016:~/cart_catkin_ws$ source /opt/ros/indigo/setup.bash
ubuntutecnalia@UbuntuTecnalia-2016:~/cart_catkin_ws$ catkin_make_isolated --install --use-ninja
Base path: /home/ubuntutecnalia/cart_catkin_ws
Source space: /home/ubuntutecnalia/cart_catkin_ws/src
Build space: /home/ubuntutecnalia/cart_catkin_ws/build_isolated
Devel space: /home/ubuntutecnalia/cart_catkin_ws/devel_isolated
Install space: /home/ubuntutecnalia/cart_catkin_ws/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 5 packages in topological order:
~~  - ceres_solver (plain cmake)
~~  - cartographer (plain cmake)
~~  - cartographer_ros_msgs
~~  - cartographer_ros
~~  - cartographer_rviz
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

==> Processing plain cmake package: 'ceres_solver'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/ubuntutecnalia/cart_catkin_ws/build_isolated/ceres_solver/install'
ninja: no work to do.
==> ninja -j8 -l8 in '/home/ubuntutecnalia/cart_catkin_ws/build_isolated/ceres_solver/install'
[1/5] Performing update step for 'ceres_src'
==> ninja install in '/home/ubuntutecnalia/cart_catkin_ws/build_isolated/ceres_solver/install'
[2/2] Install the project...
-- Install configuration: ""
<== Finished processing package [1 of 5]: 'ceres_solver'

==> Processing plain cmake package: 'cartographer'
==> Building with env: '/home/ubuntutecnalia/cart_catkin_ws/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/ubuntutecnalia/cart_catkin_ws/build_isolated/cartographer/install'
[1/1] Re-running CMake...
FAILED: /usr/bin/cmake -H/home/ubuntutecnalia/cart_catkin_ws/src/cartographer -B/home/ubuntutecnalia/cart_catkin_ws/build_isolated/cartographer/install
-- Build type: Release
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   iostreams
-- Found required Ceres dependency: Eigen version 3.2.0 in /usr/include/eigen3
-- Found required Ceres dependency: Glog in /usr/include
-- Found Ceres version: 1.11.0 installed in: /home/ubuntutecnalia/cart_catkin_ws/install_isolated
CMake Error at CMakeLists.txt:32 (find_package):
  By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen3", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/ubuntutecnalia/cart_catkin_ws/build_isolated/cartographer/install/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntutecnalia/cart_catkin_ws/build_isolated/cartographer/install/CMakeFiles/CMakeError.log".
ninja: error: rebuilding 'build.ninja': subcommand failed
<== Failed to process package 'cartographer': 
  Command '['/home/ubuntutecnalia/cart_catkin_ws/install_isolated/env.sh', 'ninja', 'build.ninja']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/ubuntutecnalia/cart_catkin_ws/build_isolated/cartographer && /home/ubuntutecnalia/cart_catkin_ws/install_isolated/env.sh ninja build.ninja

Command failed, exiting.
ubuntutecnalia@UbuntuTecnalia-2016:~/cart_catkin_ws$ 

```

I did not make the installation steps because i already have indigo installed, sourced it and try to compile and had the same error, like i told you ill keep trying to fix this by myself but i could use some help, i really want to make this work because i'm working right now with a VLP-16 for my thesis about SLAM for automated vehicles in structured environments, and this cartographer algorithms looks awesome.

Thanks again.

Regards.
David.

SirVer, 2016-11-18:
In your steps, you will have used a mix of your old workspace and the system wide installation, since the workspace will have remember it. I understand you saying that you did try from scratch, but that is not the output you posted. Please do it again and removing and restarting your workspace, just to give me confidence.

Also, could you check the output of `dpkg -L libeigen3-dev| grep cmake`. For me it lists

~~~
/usr/share/cmake-2.8
/usr/share/cmake-2.8/Modules
/usr/share/cmake-2.8/Modules/FindEigen3.cmake
~~~

alvarezmdavid, 2016-11-18:
Sorry, i get that i should remove all the workspace and reinstall it from scratch, didnt do that, sorry i'll do it, but do you mean i should remove ROS Indigo and reinstall it too? 

`ubuntutecnalia@UbuntuTecnalia-2016:~/cart_catkin_ws$ dpkg -L libeigen3-dev| grep cmake
/usr/share/cmake-2.8
/usr/share/cmake-2.8/Modules
/usr/share/cmake-2.8/Modules/FindEigen3.cmake
`

Lists the same for me but i found i have also a `/usr/share/cmake-3.2/` folder, where FindEigen3.cmake is not listed (don't have that file) i don't now if i should remove it, should i?

Ok, i erase the workspace and i did all over again

`ubuntutecnalia@UbuntuTecnalia-2016:~$ sudo apt-get update`

All worked well, then: 

`ubuntutecnalia@UbuntuTecnalia-2016:~$ sudo apt-get install -y python-wstool python-rosdep ninja-build
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
ninja-build ya está en su versión más reciente.
python-rosdep ya está en su versión más reciente.
python-wstool ya está en su versión más reciente.
0 actualizados, 0 se instalarán, 0 para eliminar y 6 no actualizados.
`

i upgraded 

`ubuntutecnalia@UbuntuTecnalia-2016:~$ sudo apt-get upgrade -y python-wstool python-rosdep ninja-build
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
Calculando la actualización... Listo
ninja-build ya está en su versión más reciente.
python-rosdep ya está en su versión más reciente.
python-wstool ya está en su versión más reciente.
Se actualizarán los siguientes paquetes:
  gstreamer1.0-plugins-bad gstreamer1.0-plugins-bad-faad
  gstreamer1.0-plugins-bad-videoparsers klibc-utils
  libgstreamer-plugins-bad1.0-0 libklibc
6 actualizados, 0 se instalarán, 0 para eliminar y 0 no actualizados.
Necesito descargar 1.372 kB de archivos.
Se utilizarán 2.048 B de espacio de disco adicional después de esta operación.
Des:1 http://es.archive.ubuntu.com/ubuntu/ trusty-updates/universe gstreamer1.0-plugins-bad-videoparsers amd64 1.2.4-1~ubuntu1.1 [61,2 kB]
Des:2 http://es.archive.ubuntu.com/ubuntu/ trusty-updates/universe gstreamer1.0-plugins-bad amd64 1.2.4-1~ubuntu1.1 [1.025 kB]
Des:3 http://es.archive.ubuntu.com/ubuntu/ trusty-updates/universe gstreamer1.0-plugins-bad-faad amd64 1.2.4-1~ubuntu1.1 [21,1 kB]
Des:4 http://es.archive.ubuntu.com/ubuntu/ trusty-updates/universe libgstreamer-plugins-bad1.0-0 amd64 1.2.4-1~ubuntu1.1 [117 kB]
Des:5 http://es.archive.ubuntu.com/ubuntu/ trusty-updates/main klibc-utils amd64 2.0.3-0ubuntu1.14.04.2 [107 kB]
Des:6 http://es.archive.ubuntu.com/ubuntu/ trusty-updates/main libklibc amd64 2.0.3-0ubuntu1.14.04.2 [40,6 kB]
Descargados 1.372 kB en 2seg. (591 kB/s)
(Leyendo la base de datos ... 280310 ficheros o directorios instalados actualmente.)
Preparando para desempaquetar .../gstreamer1.0-plugins-bad-videoparsers_1.2.4-1~ubuntu1.1_amd64.deb ...
Desempaquetando gstreamer1.0-plugins-bad-videoparsers:amd64 (1.2.4-1~ubuntu1.1) sobre (1.2.4-1~ubuntu1) ...
Preparando para desempaquetar .../gstreamer1.0-plugins-bad_1.2.4-1~ubuntu1.1_amd64.deb ...
Desempaquetando gstreamer1.0-plugins-bad:amd64 (1.2.4-1~ubuntu1.1) sobre (1.2.4-1~ubuntu1) ...
Preparando para desempaquetar .../gstreamer1.0-plugins-bad-faad_1.2.4-1~ubuntu1.1_amd64.deb ...
Desempaquetando gstreamer1.0-plugins-bad-faad:amd64 (1.2.4-1~ubuntu1.1) sobre (1.2.4-1~ubuntu1) ...
Preparando para desempaquetar .../libgstreamer-plugins-bad1.0-0_1.2.4-1~ubuntu1.1_amd64.deb ...
Desempaquetando libgstreamer-plugins-bad1.0-0:amd64 (1.2.4-1~ubuntu1.1) sobre (1.2.4-1~ubuntu1) ...
Preparando para desempaquetar .../klibc-utils_2.0.3-0ubuntu1.14.04.2_amd64.deb ...
Desempaquetando klibc-utils (2.0.3-0ubuntu1.14.04.2) sobre (2.0.3-0ubuntu1.14.04.1) ...
Preparando para desempaquetar .../libklibc_2.0.3-0ubuntu1.14.04.2_amd64.deb ...
Desempaquetando libklibc (2.0.3-0ubuntu1.14.04.2) sobre (2.0.3-0ubuntu1.14.04.1) ...
Configurando libgstreamer-plugins-bad1.0-0:amd64 (1.2.4-1~ubuntu1.1) ...
Configurando gstreamer1.0-plugins-bad-videoparsers:amd64 (1.2.4-1~ubuntu1.1) ...
Configurando gstreamer1.0-plugins-bad-faad:amd64 (1.2.4-1~ubuntu1.1) ...
Configurando gstreamer1.0-plugins-bad:amd64 (1.2.4-1~ubuntu1.1) ...
Configurando libklibc (2.0.3-0ubuntu1.14.04.2) ...
Configurando klibc-utils (2.0.3-0ubuntu1.14.04.2) ...
Procesando disparadores para libc-bin (2.19-0ubuntu6.9) ...
[... 472 lines cut ...]

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/ubuntutecnalia/CartWS/build_isolated/cartographer/install/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntutecnalia/CartWS/build_isolated/cartographer/install/CMakeFiles/CMakeError.log".
<== Failed to process package 'cartographer': 
  Command '['/home/ubuntutecnalia/CartWS/install_isolated/env.sh', 'cmake', '/home/ubuntutecnalia/CartWS/src/cartographer', '-DCMAKE_INSTALL_PREFIX=/home/ubuntutecnalia/CartWS/install_isolated', '-G', 'Ninja']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/ubuntutecnalia/CartWS/build_isolated/cartographer && /home/ubuntutecnalia/CartWS/install_isolated/env.sh cmake /home/ubuntutecnalia/CartWS/src/cartographer -DCMAKE_INSTALL_PREFIX=/home/ubuntutecnalia/CartWS/install_isolated -G Ninja

Command failed, exiting.

`

Can it be a problem about the CMake3 that i have installed?

[image]

Regards.
David.

SirVer, 2016-11-18:
Thanks for the output!

I think we have the problem now though: you have two versions of cmake installed - and only one has a FindEigen file. Uninstall cmake-3.2 and make sure that you are always using cmake-2.8 and it should work.

alvarezmdavid, 2016-11-18:
Well... apprently nop.... did all over again and.... 

`ubuntutecnalia@UbuntuTecnalia-2016:~/CarWS$ sudo rosdep install --from-paths src --ignore-src --rosdistro=indigo -y
executing command [sudo -H apt-get install -y ros-indigo-tf2-eigen]
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
Se instalarán los siguientes paquetes NUEVOS:
  ros-indigo-tf2-eigen
0 actualizados, 1 se instalarán, 0 para eliminar y 0 no actualizados.
Se necesita descargar 0 B/5.706 B de archivos.
Se utilizarán 79,9 kB de espacio de disco adicional después de esta operación.
Seleccionando el paquete ros-indigo-tf2-eigen previamente no seleccionado.
(Leyendo la base de datos ... 280173 ficheros o directorios instalados actualmente.)
Preparando para desempaquetar .../ros-indigo-tf2-eigen_0.5.13-0trusty-20160627-192052-0700_amd64.deb ...
Desempaquetando ros-indigo-tf2-eigen (0.5.13-0trusty-20160627-192052-0700) ...
Configurando ros-indigo-tf2-eigen (0.5.13-0trusty-20160627-192052-0700) ...
#All required rosdeps installed successfully
ubuntutecnalia@UbuntuTecnalia-2016:~/CarWS$ catkin_make_isolated --install --use-ninja
Base path: /home/ubuntutecnalia/CarWS
Source space: /home/ubuntutecnalia/CarWS/src
Build space: /home/ubuntutecnalia/CarWS/build_isolated
Devel space: /home/ubuntutecnalia/CarWS/devel_isolated
Install space: /home/ubuntutecnalia/CarWS/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 5 packages in topological order:
~~  - ceres_solver (plain cmake)
~~  - cartographer (plain cmake)
~~  - cartographer_ros_msgs
~~  - cartographer_ros
~~  - cartographer_rviz
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
The packages or cmake arguments have changed, forcing cmake invocation

==> Processing plain cmake package: 'ceres_solver'
==> Creating build directory: 'build_isolated/ceres_solver/install'
==> cmake /home/ubuntutecnalia/CarWS/src/cartographer_ros/ceres_solver -DCMAKE_INSTALL_PREFIX=/home/ubuntutecnalia/CarWS/install_isolated -G Ninja in '/home/ubuntutecnalia/CarWS/build_isolated/ceres_solver/install'
CMake Error: Could not find CMAKE_ROOT !!!
CMake has most likely not been installed correctly.
Modules directory not found in
/usr/share/cmake-3.2
CMake Error: Error executing cmake::LoadCache(). Aborting.

<== Failed to process package 'ceres_solver': 
  Command '['cmake', '/home/ubuntutecnalia/CarWS/src/cartographer_ros/ceres_solver', '-DCMAKE_INSTALL_PREFIX=/home/ubuntutecnalia/CarWS/install_isolated', '-G', 'Ninja']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/ubuntutecnalia/CarWS/build_isolated/ceres_solver && cmake /home/ubuntutecnalia/CarWS/src/cartographer_ros/ceres_solver -DCMAKE_INSTALL_PREFIX=/home/ubuntutecnalia/CarWS/install_isolated -G Ninja

Command failed, exiting.
`

I'm sorry i might being annoying

SirVer, 2016-11-18:
The output shows you are still using cmake 3.2. What does `which cmake` print? what about `cmake --version`.

alvarezmdavid, 2016-11-21:
Holger, Good Morning.

The output for `which cmake` shows me this:

`ubuntutecnalia@UbuntuTecnalia-2016:~$ which cmake
/usr/bin/cmake
`

`cmake -- version` shows:

`ubuntutecnalia@UbuntuTecnalia-2016:~/cart_catkin_ws$ cmake --version
cmake version 3.2.2

`

honestly i'm not getting whats going on, i removed everything (including ROS-indigo) and reinstalled it, i dont have the cmake-3.2 folder on `/usr/share/` folder 

Thanks again.

Regards.
David

SirVer, 2016-11-21:
I think you probably installed cmake-3.2 some time in the past, probably not through apt-get, but by building and installing it from source or so. And now your system partially uses cmake-3.2 (the binary at least). Try killing everything related to cmake-3.2 on your system and you should be good to go. Alternatively, reinstall your system and do not install cmake 3.2. Or you use our docker files and run Cartographer from inside a docker container. Good luck!

Closing, as this is not Cartographer related anymore.

alvarezmdavid, 2016-11-21:
Thank you, Holger.

David.

wallEVA96, 2017-05-05:
@gongyue666   hey.   I have the same issue like Urs. could U tell me how to solve this. My PC is also i386.  thank u.

mithunvarma, 2017-08-20:
Even I had the same problem. For another project I had to upgrade cmake to 3.2, so even I had both the versions and thank you guys in figuring out the problem.

Instead I just did this one and it solved my problem

` sudo cp /usr/share/cmake-2.8/Modules/FindEigen3.cmake /usr/share/cmake-3.2/Modules/ `

wallEVA96, 2017-08-20:
Ur mean is that I should upgrade the Cmake？ @mithunvarma

mithunvarma, 2017-08-20:
Nope 
Check your cmake --version and if you have cmake 3.2

As sirver mentioned 

Also, could you check the output of dpkg -L libeigen3-dev| grep cmake. For me it lists

/usr/share/cmake-2.8
/usr/share/cmake-2.8/Modules
/usr/share/cmake-2.8/Modules/FindEigen3.cmake

If you find FindEigen3.cmake file in cmake-2.8 instead of cmake-3.2 just copy it to cmake-3.2 using the command in my previous comment.(only if cmake you are using is 3.2 or something different than 2.8)

mithunvarma, 2017-08-20:
Incase you have cmake 2.8 no need to upgrade it. (Don't worry sudo apt-get upgrade doesn't automatically upgrade your cmake ) .
Just install the package for eigen ver 3 as someone mentioned above.

wallEVA96, 2017-08-20:
@mithunvarma thank u for ur suggestion, let me try it

MilanMichael, 2017-12-17:
in CmakeList.txt just replace #find_package(Eigen3 REQUIRED) with the following 
find_package(Eigen 3 QUIET)
if(NOT Eigen_FOUND)
  include(/usr/share/cmake-2.8/Modules/FindEigen3.cmake)
  set(Eigen_INCLUDE_DIR ${EIGEN3_INCLUDE_DIR})
endif()
include_directories(${Eigen_INCLUDE_DIR})

it worked for me