cartographer-project/cartographer #53: Add documentation for building and running in Docker.

damonkohler, 2016-10-13:


kurtfairfield, 2018-02-14:
@damonkohler hello, I am doing a software architecture project at UVic in BC, Canada. We are using Cartographer as our reference open source project and reviewing portions of the architecture. One of our deliverables is to submit a pull request that improves the project.
I was wondering if this is still an open issue; building documentation and running in Docker?

NHarmie, 2018-04-07:
Hello, I am part of @kurtfairfield 's team and we created a small instruction guide on building and running Cartographer in a Docker container. Please let us know if this is something that is helpful towards this issue.

[DockerInstructions.pdf](https://github.com/googlecartographer/cartographer/files/1886075/DockerInstructions.pdf)

xjtao1314, 2018-05-03:
I finally figure out how to build in docker , my steps as follows:
1. cd cartographer-master   : into cartographer main directory
2.cp Dockerfile.xenial   Dockerfile  : for me suitable as I am using ubuntu 16.04
3. docker build -t  carto_docker ./  : start install C++ env  and cartographer dependency in  docker
note : i change my dns to 8.8.8.8 before step 3, which is google  dns server