cartographer-project/cartographer #67: install error about gtest

onlytailei, 2016-10-17:
I'm installing cartographer_ros in ubuntu 14.04 and ROS indigo.
gcc version is 4.8.
Through the instruction of 
https://google-cartographer-ros.readthedocs.io/en/latest/
I build a ros-workspace named cartographer_ws(Cause I already have a workspace named catkin_ws)
When I try to execute 
`catkin_make_isolated --install --use-ninja`

I got output like that:

```
==> Processing catkin package: 'cartographer_ros'
==> Building with env: '/home/tyler/cartographer_ws/install_isolated/env.sh'
==> cmake /home/tyler/cartographer_ws/src/cartographer_ros/cartographer_ros -DCATKIN_DEVEL_PREFIX=/home/tyler/cartographer_ws/devel_isolated/cartographer_ros -DCMAKE_INSTALL_PREFIX=/home/tyler/cartographer_ws/install_isolated -G Ninja in '/home/tyler/cartographer_ws/build_isolated/cartographer_ros'
-- Found required Ceres dependency: Eigen version 3.2.8 in /usr/local/include/eigen3
-- Found required Ceres dependency: Glog in /usr/include
-- Found Ceres version: 1.11.0 installed in: /home/tyler/cartographer_ws/install_isolated
-- Build type: Release
-- Using CATKIN_DEVEL_PREFIX: /home/tyler/cartographer_ws/devel_isolated/cartographer_ros
-- Using CMAKE_PREFIX_PATH: /home/tyler/cartographer_ws/install_isolated;/opt/ros/indigo
-- This workspace overlays: /home/tyler/cartographer_ws/install_isolated;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/tyler/cartographer_ws/build_isolated/cartographer_ros/test_results
-- Found gtest: gtests will be built
CMake Error at /opt/ros/indigo/share/catkin/cmake/test/gtest.cmake:154 (add_library):
  add_library cannot create imported target "gtest" because another target
  with the same name already exists.
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/all.cmake:147 (include)
  /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:36 (find_package)

CMake Error at /opt/ros/indigo/share/catkin/cmake/test/gtest.cmake:156 (add_library):
  add_library cannot create imported target "gtest_main" because another
  target with the same name already exists.
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/all.cmake:147 (include)
  /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:36 (find_package)

-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- looking for PCL_COMMON
-- looking for PCL_OCTREE
-- looking for PCL_IO
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   iostreams
-- Configuring incomplete, errors occurred!
See also "/home/tyler/cartographer_ws/build_isolated/cartographer_ros/CMakeFiles/CMakeOutput.log".
See also "/home/tyler/cartographer_ws/build_isolated/cartographer_ros/CMakeFiles/CMakeError.log".
<== Failed to process package 'cartographer_ros': 
  Command '['/home/tyler/cartographer_ws/install_isolated/env.sh', 'cmake', '/home/tyler/cartographer_ws/src/cartographer_ros/cartographer_ros', '-DCATKIN_DEVEL_PREFIX=/home/tyler/cartographer_ws/devel_isolated/cartographer_ros', '-DCMAKE_INSTALL_PREFIX=/home/tyler/cartographer_ws/install_isolated', '-G', 'Ninja']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/tyler/cartographer_ws/build_isolated/cartographer_ros && /home/tyler/cartographer_ws/install_isolated/env.sh cmake /home/tyler/cartographer_ws/src/cartographer_ros/cartographer_ros -DCATKIN_DEVEL_PREFIX=/home/tyler/cartographer_ws/devel_isolated/cartographer_ros -DCMAKE_INSTALL_PREFIX=/home/tyler/cartographer_ws/install_isolated -G Ninja

Command failed, exiting.
```

When I directly install in in the catkin_ws, I got the same error.

SirVer, 2016-10-17:
Did you manually drop a gtest .so or .a file into your library paths?

onlytailei, 2016-10-17:
@SirVer 
There is a libgtest.a in the /usr/lib.
But even if I remove it from the /usr/lib.
I still got the same error.

SirVer, 2016-10-17:
This could be related to https://github.com/ros/catkin/issues/830, maybe this goes away if we do not call `catkin_add_gtest`.

onlytailei, 2016-10-17:
So should I comment this function about add_gtest in CMake?

SirVer, 2016-10-17:
That might fix the issue for now - You can also comment the `google_test()` function in the CMakeLists.txt one directory further down to disable testing.

onlytailei, 2016-10-17:
```
/home/tyler/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/node_main.cc:167:28: error: expected ';' at end of declaration list
  bool terminating_ = false GUARDED_BY(mutex_);
```

Another problem accrued...

SirVer, 2016-10-17:
Seems like an incompatibility in the compiler. GUARDED_BY seems to not be found. 

Please set the environment variable `export VERBOSE=1` and rerun the full installation from scratch, capturing the output and post it in a gist.

onlytailei, 2016-10-17:
```
<== Finished processing package [3 of 5]: 'cartographer_ros_msgs'

==> Processing catkin package: 'cartographer_ros'
==> Building with env: '/home/tyler/cartographer_ws/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/tyler/cartographer_ws/build_isolated/cartographer_ros'
ninja: no work to do.
==> ninja -j8 -l8 in '/home/tyler/cartographer_ws/build_isolated/cartographer_ros'
[1/2] Building CXX object cartographer_ros/CMakeFiles/cartographer_node.dir/node_main.cc.o
FAILED: /usr/bin/c++   -DDISABLE_PCAP -DDISABLE_PNG -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -Wno-deprecated -O3 -DNDEBUG -isystem /usr/include/vtk-5.8 -isystem /usr/local/include/eigen3 -isystem /home/tyler/cartographer_ws/install_isolated/include -isystem /opt/ros/indigo/include -isystem /usr/include/eigen3 -isystem /usr/include/lua5.2 -I. -I/home/tyler/cartographer_ws/src/cartographer_ros/cartographer_ros -isystem /usr/include/pcl-1.7 -isystem /usr/include/ni -isystem /usr/include/openni2     -pthread -std=c++11  -Wall -Wpedantic -Werror=format-security -Werror=reorder -Werror=return-type -Werror=uninitialized -O3 -DNDEBUG -MMD -MT cartographer_ros/CMakeFiles/cartographer_node.dir/node_main.cc.o -MF "cartographer_ros/CMakeFiles/cartographer_node.dir/node_main.cc.o.d" -o cartographer_ros/CMakeFiles/cartographer_node.dir/node_main.cc.o -c /home/tyler/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/node_main.cc
/home/tyler/cartographer_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/node_main.cc:167:28: error: expected ';' at end of declaration list
  bool terminating_ = false GUARDED_BY(mutex_);
                           ^
1 error generated.
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer_ros': 
  Command '['/home/tyler/cartographer_ws/install_isolated/env.sh', 'ninja', '-j8', '-l8']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/tyler/cartographer_ws/build_isolated/cartographer_ros && /home/tyler/cartographer_ws/install_isolated/env.sh ninja -j8 -l8

Command failed, exiting.

```

onlytailei, 2016-10-17:
This seems good
`  SensorBridge sensor_bridge_ GUARDED_BY(mutex_);`
But this is an error
`  bool terminating_ = false GUARDED_BY(mutex_);`

SirVer, 2016-10-17:
What does ` /usr/bin/c++ --version` output?

onlytailei, 2016-10-17:
Quite wired. 
it linked to clang3.7
Relink it to g++, now everything works.

Conclude that:
comment line 34 
`google_enable_tesing()`
in src/cartographer_ros/cartographer_ros/CMakeLists.txt

Thank you very much @SirVer  
Really help a lot.

onlytailei, 2016-10-17:
When running 2d demo:

`[ WARN] [1476713426.029530762, 1432647016.954832117]: W1017 22:10:26.000000 11690 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.963997200 but the earliest data is at time 1432647016.982176773, when looking up transform from frame [imu_link] to frame [base_link]
[ WARN] [1476713426.035637161, 1432647016.964886038]: W1017 22:10:26.000000 11690 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.970029900 but the earliest data is at time 1432647016.982176773, when looking up transform from frame [imu_link] to frame [base_link]
[ WARN] [1476713426.037167923, 1432647016.964886038]: W1017 22:10:26.000000 11690 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.971389900 but the earliest data is at time 1432647016.982176773, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
[ WARN] [1476713426.041435726, 1432647016.974951963]: W1017 22:10:26.000000 11690 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.975897100 but the earliest data is at time 1432647016.982176773, when looking up transform from frame [imu_link] to frame [base_link]
[ WARN] [1476713426.041571208, 1432647016.974951963]: W1017 22:10:26.000000 11690 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.975931000 but the earliest data is at time 1432647016.982176773, when looking up transform from frame [imu_link] to frame [base_link]
[ WARN] [1476713426.047592037, 1432647016.974951963]: W1017 22:10:26.000000 11690 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.982035800 but the earliest data is at time 1432647016.982176773, when looking up transform from frame [imu_link] to frame [base_link]
[cartographer_node-3] process has died [pid 11690, exit code -8, cmd /home/tyler/cartographer_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/tyler/cartographer_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua echoes:=horizontal_laser_2d __name:=cartographer_node __log:=/home/tyler/.ros/log/72a823ea-9473-11e6-b439-00216a8ea37a/cartographer_node-3.log].
log file: /home/tyler/.ros/log/72a823ea-9473-11e6-b439-00216a8ea37a/cartographer_node-3*.log
`

How can I debug that?

SirVer, 2016-10-17:
Does the demo work? I think this is due to the sensor data arriving at different times - We do not have a pose information before the first scan match, so any data before (like in this case from the IMU) can be transformed yet. So this is working as intended.

onlytailei, 2016-10-17:
No, it does not work.
No map output.

SirVer, 2016-10-20:
From the output above it is hard to know where the process died. Is there not more output - like a LOG(FATAL) somewhere?

I suggest you try running our docker image and see if everything works with that.

SirVer, 2016-10-26:
I cannot reproduce this. Using a pristine docker container based on `ros:indigo` and following [the ros installation instructions](https://google-cartographer-ros.readthedocs.io/en/latest/?badge=latest#building-installation) everything compiles and starts cleanly on my system. 

For running the demo, I needed to install `ros-indigo-robot-state-publisher` - could that be the problem you have and you just missed the error output because of all the warnings?

Closing since I cannot reproduce. The gtest issues is tracked in https://github.com/googlecartographer/cartographer/issues/73.

closed 2016-10-26 by SirVer

snehagn, 2017-10-31:
I had the same error and I commented out line 34 in in src/cartographer_ros/cartographer_ros/CMakeLists.txt. The program compiled till 4/5 and then gave me this error. 

CMake Error at /opt/ros/kinetic/share/catkin/cmake/assert.cmake:3 (message):
  

  Assertion failed: GTEST_LIBRARIES (value is '')

Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/test/gtest.cmake:84 (assert)
  /opt/ros/kinetic/share/catkin/cmake/test/gtest.cmake:34 (catkin_add_executable_with_gtest)
  CMakeLists.txt:139 (catkin_add_gtest)

-- Configuring incomplete, errors occurred!
See also "/home/udoo2/google_carto/build_isolated/cartographer_ros/CMakeFiles/CMakeOutput.log".
See also "/home/udoo2/google_carto/build_isolated/cartographer_ros/CMakeFiles/CMakeError.log".
<== Failed to process package 'cartographer_ros': 
  Command '['/home/udoo2/google_carto/install_isolated/env.sh', 'cmake', '/home/udoo2/google_carto/src/cartographer_ros/cartographer_ros', '-DCATKIN_DEVEL_PREFIX=/home/udoo2/google_carto/devel_isolated/cartographer_ros', '-DCMAKE_INSTALL_PREFIX=/home/udoo2/google_carto/install_isolated', '-G', 'Ninja']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/udoo2/google_carto/build_isolated/cartographer_ros && /home/udoo2/google_carto/install_isolated/env.sh cmake /home/udoo2/google_carto/src/cartographer_ros/cartographer_ros -DCATKIN_DEVEL_PREFIX=/home/udoo2/google_carto/devel_isolated/cartographer_ros -DCMAKE_INSTALL_PREFIX=/home/udoo2/google_carto/install_isolated -G Ninja

Command failed, exiting.

Are there more lines I would need to comment out and from which files? Any help is appreciated thanks!

gaschler, 2017-11-02:
To me it looks like you have a non-default version of gtest or in your library path.

More details are needed here.
- recompile with `export VERBOSE=1` (as was already suggested)
- gcc --version
- Contents of (install|devel)_isolated/cartographer/devel/CMakeCache.txt
- env | grep ROS

As always, perhaps you already did this, please retry with a current checkout from git and compile again after deleting all build/install folders.

Of course, you could work around by commenting out `catkin_add_gtest`, as it was suggested, but Ubuntu 16.04 is fully supported, there is no need for a work around.

RuochenYin, 2017-11-09:
I got the same problem as @snehagn 
I'm installing cartographer_ros in ubuntu 14.04 and ROS indigo.
gcc version is 4.8, and I also commented out line 34 in in src/cartographer_ros/cartographer_ros/CMakeLists.txt

CMake Error at /opt/ros/indigo/share/catkin/cmake/assert.cmake:3 (message):
  

  Assertion failed: GTEST_LIBRARIES (value is 'GMOCK_LIBRARIES-NOTFOUND')

Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/test/gtest.cmake:84 (assert)
  /opt/ros/indigo/share/catkin/cmake/test/gtest.cmake:34 (catkin_add_executable_with_gtest)
  CMakeLists.txt:139 (catkin_add_gtest)

-- Configuring incomplete, errors occurred!
See also "/home/roland/catkin_ws/build_isolated/cartographer_ros/CMakeFiles/CMakeOutput.log".
See also "/home/roland/catkin_ws/build_isolated/cartographer_ros/CMakeFiles/CMakeError.log".
ninja: error: rebuilding 'build.ninja': subcommand failed
<== Failed to process package 'cartographer_ros': 
  Command '['/home/roland/catkin_ws/install_isolated/env.sh', 'ninja', 'build.ninja']' returned non-zero exit status 1

Reproduce this error by running==> cd /home/roland/catkin_ws/build_isolated/cartographer_ros && /home/roland/catkin_ws/install_isolated/env.sh ninja build.ninja

Command failed, exiting.

And I already set the environment variable export VERBOSE=1.  But still it comes out anyway.
So if any idea can solve this problem, please let me know, thanks!

SirVer, 2017-11-09:
@RuochenYin Please open a new issue and provide the full output of all commands you ran to set up your catkin workstation.