cartographer-project/cartographer #118: Problems with GMock

af-silva, 2016-10-31:
Version of GMock 1.6.0+svn437-0ubuntu5 (installed from the repos)
Version of Ubuntu 14.04

Please refer to the error below. This seems to be happening since four days ago?? Can it be a problem with the GMock version that is being used? the last version that I can find at [link](https://github.com/google/googletest/releases) is the 1.8 version... the code already compiled with success before, I will try to revert this last commits to verify if the error persists, if the error happen because the version of GMock installed please update the minimum version needed to compile the project.

Many thanks in advance if some one can help with this problems.
And great work guys :) 

`In file included from /usr/include/gmock/gmock-actions.h:46:0,
                 from /usr/include/gmock/gmock.h:58,
                 from /home/andre/RosWorkspaces/catkin_ws/src/cartographer/cartographer/transform/rigid_transform_test_helpers.h:26,
                 from /home/andre/RosWorkspaces/catkin_ws/src/cartographer/cartographer/kalman_filter/pose_tracker_test.cc:25:
/usr/include/gmock/internal/gmock-internal-utils.h: In static member function ‘static testing::internal::StlContainerView<Element [N]>::const_reference testing::internal::StlContainerView<Element [N]>::ConstReference(const Element (&)[N])’:
/usr/include/gmock/internal/gmock-internal-utils.h:439:27: error: ‘kReference’ was not declared in this scope
     return type(array, N, kReference);
                           ^
/usr/include/gmock/internal/gmock-internal-utils.h: In static member function ‘static testing::internal::StlContainerView<Element [N]>::type testing::internal::StlContainerView<Element [N]>::Copy(const Element (&)[N])’:
/usr/include/gmock/internal/gmock-internal-utils.h:446:27: error: ‘kCopy’ was not declared in this scope
     return type(array, N, kCopy);
                           ^
/usr/include/gmock/internal/gmock-internal-utils.h: In static member function ‘static testing::internal::StlContainerView<std::tuple<_T1, _T2> >::const_reference testing::internal::StlContainerView<std::tuple<_T1, _T2> >::ConstReference(const std::tuple<_T1, _T2>&)’:
/usr/include/gmock/internal/gmock-internal-utils.h:464:47: error: ‘kReference’ was not declared in this scope
     return type(get<0>(array), get<1>(array), kReference);
                                               ^
/usr/include/gmock/internal/gmock-internal-utils.h: In static member function ‘static testing::internal::StlContainerView<std::tuple<_T1, _T2> >::type testing::internal::StlContainerView<std::tuple<_T1, _T2> >::Copy(const std::tuple<_T1, _T2>&)’:
/usr/include/gmock/internal/gmock-internal-utils.h:468:47: error: ‘kCopy’ was not declared in this scope
     return type(get<0>(array), get<1>(array), kCopy);
                                               ^
In file included from /home/andre/RosWorkspaces/catkin_ws/src/cartographer/cartographer/kalman_filter/pose_tracker_test.cc:22:0:
/home/andre/RosWorkspaces/catkin_ws/src/cartographer/cartographer/common/lua_parameter_dictionary_test_helpers.h: In member function ‘virtual std::string cartographer::common::DummyFileResolver::GetFileContentOrDie(const string&)’:
/home/andre/RosWorkspaces/catkin_ws/src/cartographer/cartographer/common/lua_parameter_dictionary_test_helpers.h:42:3: error: control reaches end of non-void function [-Werror=return-type]
   }
   ^
/home/andre/RosWorkspaces/catkin_ws/src/cartographer/cartographer/common/lua_parameter_dictionary_test_helpers.h: In member function ‘virtual std::string cartographer::common::DummyFileResolver::GetFullPathOrDie(const string&)’:
/home/andre/RosWorkspaces/catkin_ws/src/cartographer/cartographer/common/lua_parameter_dictionary_test_helpers.h:46:3: error: control reaches end of non-void function [-Werror=return-type]
   }
   ^
cc1plus: some warnings being treated as errors
make[2]: *** [cartographer/kalman_filter/CMakeFiles/kalman_filter_pose_tracker_test.dir/pose_tracker_test.cc.o] Error 1
make[1]: *** [cartographer/kalman_filter/CMakeFiles/kalman_filter_pose_tracker_test.dir/all] Error 2
make: *** [all] Error 2
andre@prtandre:~/RosWorkspaces/catkin_ws/src/cartographer/build$ sudo apt-cache google-mock
[sudo] password for andre: 
andre@prtandre:~/RosWorkspaces/catkin_ws/src/cartographer/build$ apt-cache google-mock
E: Invalid operation google-mock
andre@prtandre:~/RosWorkspaces/catkin_ws/src/cartographer/build$ apt-cache info google-mock
E: Invalid operation info
`

skohlbr, 2016-10-31:
Just to confirm you're not the only one, I'm getting this on 14.04, AMD64, `google-mock` installed from .debs.

af-silva, 2016-10-31:
I found a temporary fix to this, I revert to the commit `a5b81ff159d302441b33a51585506e02b137a95f` of four days ago and everything compiles without a problem. My guess is that we need to update the GMock to a more recent version 1.7 our something is missing in the FindGMock.cmake file. I don't have much time to look around, for now I will be working on top the commit a mention earlier.

SirVer, 2016-11-01:
This enum should be defined in `/usr/src/gmock/gtest/include/gtest/internal/gtest-internal.h`, line 938. Could you check that this definition is there on your system?

af-silva, 2016-11-02:
@SirVer I can confirm that the enum is declared in the header file you mention 
```
// The relation between an NativeArray object (see below) and the 
// native array it represents.
enum RelationToSource {
  kReference,  // The NativeArray references the native array.
  kCopy           // The NativeArray makes a copy of the native array and
                      // owns the copy.
};
```
Here is the snippet of the code. 

If more information is needed please ask! :) 
I also facing some problems with the google gtest in the cartographer_ros package, something weird is happening, at the moment I can't compile the project because of this error: 

`undefined reference to `testing::internal::MakeAndRegisterTestInfo`

Maybe this can help pinpoint what I'm doing wrong, again I had already the code to running in this machine.

Thanks for the help...

SirVer, 2016-11-02:
Have you tried nuking your catkin_workspace and start over a-fresh? It seems cmake mixes different versions of gmock on your system. Could it be that it is confused about changes we did to `FindGmock.cmake`?

skohlbr, 2016-11-02:
I can confirm that on my catkin_tools based workspace, a clean rebuild fixed the issue.

SirVer, 2016-11-02:
interesting. I close this issue then. If somebody still has the issue, please reopen.

closed 2016-11-02 by SirVer

ahundt, 2017-07-25:
@SirVer I've encountered this again. Which catkin workspace should be removed, the whole ros install or just the cartographer one?

Here are my build errors:
https://gist.github.com/ahundt/67191f868db0d8f506316236238f7b55

Here is my full install script, based on the read the docs instructions:
https://github.com/ahundt/robotics_setup/blob/master/cartographer_ros.sh

SirVer, 2017-07-25:
I usually do `rm -rf *install` and rebuild everything.

ahundt, 2017-07-25:
Cool, got it working earlier by blasting the cartographer install files but couldn't find this issue again. Thanks!