cartographer-project/cartographer #127: about the global loop-closure optimization between sub-maps

gogofight, 2016-11-16:
The following is 2d map I made using laser lidar :

[image]
[image]
[image]
[image]

The 1th,2th,3th maps are correct, but the 4th map has led to a global error.

What I want is that a submap has been drawn will not be affected by any new submap, and i need not global loop-closure optimization matching or correction fitting between submaps.

Because of some local similarities between adjacent sub-maps, the global loop-closure optimization algorithm results in fitting distorted error between sub-maps. But, i do not need this map optimizating or handling.

I want to know whether the global loop-closure optimization can be removed by parameters configuration and which parameters and how to do. Are they in file "[sparse_pose_graph.lua](https://github.com/googlecartographer/cartographer/blob/master/configuration_files/sparse_pose_graph.lua)" ? Is this parameter "**optimize_every_n_scans = 90**"? 

Or the fitting distorted error between sub-maps is generated from other place ? and where ?

I think the global loop-closure optimization algorithm has an not small problem and bug in practical application , because it will lead to the entire map mapping errors and useless !

I will upload my bag file, some can download it to test for datails.

damonkohler, 2016-11-17:
@gogofight I assume you resolved this issue yourself? Disabling loop closure is not a good idea. Doing so can significantly reduce the quality of mapping.

Instead, if you're having trouble with repetitive environments, it's possible that you need to tune your settings for your sensor. What sort of sensor is it? Can you share your configuration and a bag?

gogofight, 2016-11-17:
@damonkohler 

config file: https://drive.google.com/open?id=0B2-B3l8213hNa3hLczR4MGdCU0E

bag file: https://drive.google.com/open?id=0B2-B3l8213hNeFVJcE0wZzV4VmM

The following is 2d map I made using laser lidar :

[image]
[image]
[image]
[image]

The 1th,2th,3th maps are correct, but the 4th map has led to a global error.

What I want is that a submap has been drawn will not be affected by any new submap, and i need not global loop-closure optimization matching or correction fitting between submaps.

Because of some local similarities between adjacent sub-maps, the global loop-closure optimization algorithm results in fitting distorted error between sub-maps. But, i do not need this map optimizating or handling.

I want to know whether the global loop-closure optimization can be removed by parameters configuration and which parameters and how to do. Are they in file "sparse_pose_graph.lua" ?

Or the fitting distorted error between sub-maps is generated from other place ? and where ?

If you solve the problem by setting the config files, please tell me, thanks !!!

SirVer, 2016-11-17:
> I think the global loop-closure optimization algorithm has an not small problem and bug in practical application , because it will lead to the entire map mapping errors and useless !

The 2D SLAM has seen very wide application on many buildings around the world - mostly on backpacks, but also on cars and robots. Rest assure that there are no huge bugs in the system and that your problem is either a tuning or an invariance-violation issue.

For example, Map 3 does not look correct to me - it overlaps the second room. It seems to me that either some sensors are in disagreement about how the robot actually moves or that the 2D assumption (i.e. no change in altitude) is violated here.

gogofight, 2016-11-17:
@SirVer 
Ok ! There is no problem in the general scenary , such as in some small space
Could you try to test the bag file for just a detail checking ?
I will check my device sensor parameters and data. 
Thank you all the same!

damonkohler, 2016-11-18:
We ran through the bag and determined that, for your laser scanner, you'll need a higher min score and a higher huber scale to avoid using bad loop closures. So, remove the line in your configuration that disables loop closure and instead add these lines:

SPARSE_POSE_GRAPH.optimization_problem.huber_scale = 1e2
SPARSE_POSE_GRAPH.constraint_builder.min_score = 0.8

closed 2016-11-18 by damonkohler