cartographer-project/cartographer #134: running on windows?

antithing, 2016-11-18:
Has anyone got this up on windows? Is it possible?

Thanks!

SirVer, 2016-11-18:
Nobody has! You could be the first....

That said, it probably requires quite some work.

antithing, 2016-11-18:
the FIRST you say!  Sadly, i do not think my code-fu is up to it!

SirVer, 2016-11-21:
Well, with ROS not running on Windows - Linux is kinda the lingua franca in robotics. Sorry.

Closing, as windows support is not on the roadmap for now.

closed 2016-11-21 by SirVer

liuxinren456852, 2016-12-17:
I have successfully compiled in the windows platform！

antithing, 2016-12-17:
Impressed! :) Any chance that you could share/github the code?

tpengti, 2017-05-18:
I have successfully compiled in the windows platform for 2D-SLAM
https://github.com/tpengti/cartographer_windows

sunmk2006, 2017-08-16:
I have successfully compiled in the windows platform for 2D-SLAM as @tpengti ,https://github.com/tpengti/cartographer_windows,The test routine will run for more than ten minutes , but in the linux platform this routine only takes three or four seconds to run,I couldn't understand the result because the windows Library's compilation did not change any code,Can someone help me?

damonkohler, 2017-08-18:
@sunmk2006, it sounds like you compiled in debug mode or so.

sunmk2006, 2017-08-21:
@damienrg ,yes,you are right,I compiled in release mode,The cartographer worked very well! Thank you very much!

paul-michalik, 2017-08-28:
Did anybody considered merging the required changes back to master? Or were no changes required?

mmazat, 2018-02-19:
Made it easy for everyone, cartographer is now compiled and build on windows by the Microsoft Vcpkg,  https://github.com/Microsoft/vcpkg

antithing, 2018-02-20:
Thanks! Is this just the 2d slam, as in above posts? Or the 3d also?

mmazat, 2018-02-20:
it is cartographer library in full...

antithing, 2018-02-20:
Awesome! Thank you very much. I am trying to build this, with vcpkg, and it is failing with:

```
Error: Building package boost-config:x86-windows failed with: BUILD_FAILED
Please ensure you're using the latest portfiles with `.\vcpkg update`, then
submit an issue at https://github.com/Microsoft/vcpkg/issues including:
  Package: boost-config:x86-windows
  Vcpkg version: 0.0.104-unknownhash
```

What am I missing?

mmazat, 2018-02-20:
This is boost installation problem. There might be hick ups but vcpkg works, in case of issue you may report them.. my advice is to use higher version of Visual Studio (2015 or 2017 perhaps)

antithing, 2018-02-20:
I am on vs2017. I actually already have boost installed, can I just point the vcpkg to my existing files?

ras0219-msft, 2018-02-21:
We use cmake, which uses curl internally, so you'll need to set the usual curl environment variables for your proxy (`HTTPS_PROXY`)[1].

[1] https://stackoverflow.com/questions/9445489/performing-http-requests-with-curl-using-proxy

antithing, 2018-02-21:
Ah it was my firewall at the office. Another machine has it installing now. Thanks again! One more question... Is there any example / demo floating around to show how to use cartographer on windows, without ROS? If you could point me in the right direction to get started, it would be much appreciated.

anitaHey, 2021-10-06:
@antithing I also look for the direction to get started with using cartographer on windows, without ROS. Is there any experience can share? Thank you so much!