cartographer-project/cartographer #168: Cartographer build error

maxblues, 2016-12-20:
Hi!

I tried to build cartographer on my system (Ubuntu 14.04, gcc 4.8.4, ROS Indigo).
Followed the instructions for ROS, and after executing
`catkin_make_isolated --install --use-ninja`

I got the following error. Could you please help me out?  Thanks!

```
==> Processing catkin package: 'cartographer_ros'
==> Creating build directory: 'build_isolated/cartographer_ros'
==> Building with env: '/home/max/install_isolated/env.sh'
==> cmake /home/max/src/cartographer_ros/cartographer_ros -DCATKIN_DEVEL_PREFIX=/home/max/devel_isolated/cartographer_ros -DCMAKE_INSTALL_PREFIX=/home/max/install_isolated -G Ninja in '/home/max/build_isolated/cartographer_ros'
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Found required Ceres dependency: Eigen version 3.2.0 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Performing Test GFLAGS_IN_GOOGLE_NAMESPACE
-- Performing Test GFLAGS_IN_GOOGLE_NAMESPACE - Success
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.12.0 installed in: /home/max/install_isolated with components: [LAPACK, SuiteSparse, SparseLinearAlgebraLibrary, SchurSpecializations, OpenMP]
-- Build type: Release
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Found Threads: TRUE  
-- Found GMock: gmock_main;-lpthread  
-- Using CATKIN_DEVEL_PREFIX: /home/max/devel_isolated/cartographer_ros
-- Using CMAKE_PREFIX_PATH: /home/max/install_isolated;/opt/ros/indigo
-- This workspace overlays: /home/max/install_isolated;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/max/build_isolated/cartographer_ros/test_results
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- Using these message generators: gencpp;genlisp;genpy
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for one of the modules 'yaml-cpp>=0.5.1'
-- checking for module 'eigen3'
--   found eigen3, version 3.2.0
-- Found eigen: /usr/include/eigen3  
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found openni: /usr/lib/libOpenNI.so  
-- Found libusb-1.0: /usr/include  
-- looking for PCL_COMMON
-- Found PCL_COMMON: /usr/lib/libpcl_common.so  
-- looking for PCL_OCTREE
-- Found PCL_OCTREE: /usr/lib/libpcl_octree.so  
-- looking for PCL_IO
-- Found PCL_IO: /usr/lib/libpcl_io.so  
-- Found PCL: /usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libpthread.so;optimized;/usr/lib/libpcl_common.so;debug;/usr/lib/libpcl_common.so;optimized;/usr/lib/libpcl_octree.so;debug;/usr/lib/libpcl_octree.so;/usr/lib/libOpenNI.so;vtkCommon;vtkRendering;vtkHybrid;vtkCharts;optimized;/usr/lib/libpcl_io.so;debug;/usr/lib/libpcl_io.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libpthread.so;/usr/lib/libOpenNI.so;vtkCommon;vtkRendering;vtkHybrid;vtkCharts  
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "2.91.0") 
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   iostreams
CMake Error at cartographer_ros/CMakeLists.txt:1 (google_library):
  Unknown CMake command "google_library".

-- Configuring incomplete, errors occurred!
See also "/home/max/build_isolated/cartographer_ros/CMakeFiles/CMakeOutput.log".
See also "/home/max/build_isolated/cartographer_ros/CMakeFiles/CMakeError.log".
<== Failed to process package 'cartographer_ros': 
  Command '['/home/max/install_isolated/env.sh', 'cmake', '/home/max/src/cartographer_ros/cartographer_ros', '-DCATKIN_DEVEL_PREFIX=/home/max/devel_isolated/cartographer_ros', '-DCMAKE_INSTALL_PREFIX=/home/max/install_isolated', '-G', 'Ninja']' returned non-zero exit status 1

```

romainreignier, 2016-12-21:
I have build cartographer successfully before but since yesterday, I have the exact same error.

damonkohler, 2016-12-21:
Our CMake configuration has changed in all repos. Please make sure that all of your cartographer_* repos are up-to-date with master.

closed 2016-12-21 by damonkohler

romainreignier, 2016-12-21:
Actually, it is now fixed with: https://github.com/googlecartographer/cartographer_ros/pull/234