cartographer-project/cartographer #175: Online Google Cartographer with kinect v2

stevenyslin, 2016-12-26:
Is there possible that just use kinect v2 with Google Cartographer?
Because I did not see this part after searching the documents.
Thanks a lot.

jefft255, 2016-12-29:
I don't think this would easily be done at the moment. Cartographer is heavily focused on LiDAR SLAM. You can always try to publish your Kinect data as point cloud messages on ROS and see what happens.

damonkohler, 2017-01-05:
You can use any sensor that publishes PointCloud2 messages. See https://google-cartographer-ros.readthedocs.io/en/latest/ros_api.html

There are examples of using depth image data in the cartographer_turtlebot repo.

closed 2017-01-05 by damonkohler

SirVer, 2017-01-09:
@jefft255 I think your statment is incorrect - Cartographer is used successfully on the turtlebot 3 and gives good results with the depth camera it has on it.

alexsmartens, 2018-04-02:
@SirVer could you help with an example of using cartographer with kinect v2?
I tried to figure this out in the turtlebot 3 repository, but did not find any specifics other than [this open issue](https://github.com/ROBOTIS-GIT/turtlebot3/issues/159)