cartographer-project/cartographer #177: Build failure on Debian testing

nxdefiant, 2017-01-06:
Building cartographer fails on Debian testing. Googletest version is 1.8.0-6 (Debian)

Build output is as follows:

```
==> make cmake_check_build_system in '/opt/ros/ros_catkin_ws/build_isolated/cartographer/install'
-- Build type: Release
-- Found PythonInterp: /usr/bin/python (found version "2.7.13") 
CMake Error at /usr/src/googletest/googletest/cmake/internal_utils.cmake:149 (add_library):
  add_library cannot create target "gmock" because another target with the
  same name already exists.  The existing target is a static library created
  in source directory "/usr/src/googletest/googlemock".  See documentation
  for policy CMP0002 for more details.
Call Stack (most recent call first):
  /usr/src/googletest/googletest/cmake/internal_utils.cmake:172 (cxx_library_with_type)
  /usr/src/gmock/CMakeLists.txt:84 (cxx_library)

CMake Error at /usr/src/googletest/googletest/cmake/internal_utils.cmake:149 (add_library):
  add_library cannot create target "gmock_main" because another target with
  the same name already exists.  The existing target is a static library
  created in source directory "/usr/src/googletest/googlemock".  See
  documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  /usr/src/googletest/googletest/cmake/internal_utils.cmake:172 (cxx_library_with_type)
  /usr/src/gmock/CMakeLists.txt:89 (cxx_library)

CMake Error at /usr/src/gmock/CMakeLists.txt:106 (install):
  install TARGETS given target "gmock" which does not exist in this
  directory.

-- Found GMock: gmock_main;-lpthread  
-- Boost version: 1.62.0
-- Found the following Boost libraries:
--   iostreams
--   regex
-- Found required Ceres dependency: Eigen version 3.3.0 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found Ceres version: 1.12.0 installed in: /usr/local with components: [LAPACK, SchurSpecializations, OpenMP]
-- Configuring incomplete, errors occurred!
See also "/opt/ros/ros_catkin_ws/build_isolated/cartographer/install/CMakeFiles/CMakeOutput.log".
See also "/opt/ros/ros_catkin_ws/build_isolated/cartographer/install/CMakeFiles/CMakeError.log".
Makefile:3786: die Regel f?r Ziel ?cmake_check_build_system? scheiterte
make: *** [cmake_check_build_system] Fehler 1
<== Failed to process package 'cartographer': 
  Command '['/usr/local/env.sh', 'make', 'cmake_check_build_system']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /opt/ros/ros_catkin_ws/build_isolated/cartographer && /usr/local/env.sh make cmake_check_build_system

Command failed, exiting.
```

SirVer, 2017-01-09:
Could you provide the exact debian version? I am unfamiliar with debians version naming scheme and I need to know which docker container I need to use to reproduce this issue.

nxdefiant, 2017-01-09:
This is Debian testing, named [Stretch](https://wiki.debian.org/DebianStretch) which means that it is not released yet. So not sure if there is a container yet. Stretch will be released as the new stable release somewhere soon.

closed 2017-05-04 by SirVer