cartographer-project/cartographer #190: Is that possible to manually indicate loops?

nangong, 2017-01-26:
Thanks to this great work , cartographer works very well in indoor environment. 
I am current using cartographer to map a large area, but I found the loop closure does not work in such environment.

[image]

In the picture , the two branches are actually the same place. Are there any APIs for manually fixing error loop detection?

SirVer, 2017-01-26:
> I am current using cartographer to map a large area, but I found the loop closure does not work in such environment.

We mapped city scales with Cartographer no problem. It is likely that your problem is elsewhere:
Maybe you need to tune your parameters better. If you provide us with bag, urdf, launch files and your config we might be able to help. Also it seems you are building a 2D map - is your laser always level? Is your floor really flat? Do you have an IMU?

> Are there any APIs for manually fixing error loop detection?

No. This is feasible to add in the optimization problem, but it was not really necessary.

SirVer, 2017-01-30:
Closing. Please open a new bug with data + bags + config if you want tuning help.

closed 2017-01-30 by SirVer