cartographer-project/cartographer #195: Google test installer

Sladew23, 2017-01-30:
Hi,

I use Ubuntu 16.04 LTS, G++ compiler version 5.4.0. And I would like to install and use cartographer without ROS.

So after having run successfully (and consecutively) install_debs.sh and install_ceres.h (located both  inside cartographer/scripts), I executed install_cartographer.sh script 
and I got this error:
`# Build and install Cartographer.
cd cartographer
mkdir build
cd build
cmake .. -G Ninja
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at common/CMakeLists.txt:19 (google_test):
  Unknown CMake command "google_test".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.5)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
`
It seems like the install_debs.sh hasn't installed google test.

What am I supposed to do now? Install directly google test inside cartographer which I tried and it wasn' successful ?

Thanks in advance for your answer.

damonkohler, 2017-01-31:
It looks like you're somehow executing cartographer/common/CMakeLists.txt directly instead of the parent script in cartographer/. Could you provide a gist of your terminal session?

Sladew23, 2017-01-31:
Here is the gist link of my terminal session followed by some explanations about what I did: https://gist.github.com/Sladew23/07b3802f89e0cf620a8017a5b85bd884 
1st Attempt (gistfile1.txt ):
By being in the parent cartographer I run the program install_cartographer (located in cartographer/scripts/)
The terminal told me: 
"CMake Error at common/CMakeLists.txt:19 (google_test):
  Unknown CMake command "google_test" 

2nd Attempt (gistfile2.txt ):
I tried to install directly in cartographer, so I created build folder and did
`cmake .. -G Ninja` command which worked however while running `ninja` there is an error of a file that cannot be found.

damonkohler, 2017-01-31:
Are you following the installation instructions here?

https://google-cartographer.readthedocs.io/en/latest/

The install_cartographer.sh script is really for use by the Docker image. We don't recommend it for use outside of that.

It looks like perhaps Ceres is not installed properly. That would cause your build to fail.

Sladew23, 2017-02-01:
I followed the installation instructions given here: https://google-cartographer.readthedocs.io/en/latest/ (the link you suggested me).
Moreover, these (three) instructions in the link are exactly identical to the ones written respectively by install_debs.sh, install_ceres & install_cartographer.sh located in cartographer/scripts/ folder.

However, like you mentioned I installed docker (following this tutorial: https://www.digitalocean.com/community/tutorials/how-to-install-and-use-docker-getting-started) and by typing the following command lines (in /cartographer), it installed without problem:
`$ cd ~/cartographer/`
`$ cp Dockerfile.xenial Dockerfile`
`$ sudo docker build -t xenial .`

Now that it is installed, I would like to know if there is a tutorial on how to use cartographer (without ROS) ? Thanks.

damonkohler, 2017-02-02:
The scripts make assumptions about the path they're being executed from (which is the source of your original issue). Also, they are subject to change as necessary to improve the Docker build process.

The best documentation for using Cartographer without ROS is the [cartographer_ros](https://github.com/googlecartographer/cartographer_ros) implementation itself. By itself, Cartographer is only a C++ library. You'll need to wire it up to your existing infrastructure to make use of it.

closed 2017-02-02 by damonkohler