cartographer-project/cartographer #198: Error in Yaw estimation on 180 deg turns.

ajakhotia, 2017-02-07:
Hi All
    I am using Cartographer for 2D slam on a robot with a 10m range, 270 deg FOV, 15 Hz planar laser scanner and wheel encoders. It seems like ceres scan matcher is incorrectly estimating the yaw after a 180-deg turn which, when multiplied by the arm-length as the robot travels ahead, leads to a large error in the position of the robot. This error gets corrected by the pose graph matcher eventually causing big jumps in the pose of the robot. The final maps I get out of the system look pretty good but I also care about the real-time accuracy of the localization. 

As you can see in the images, the scans are in a very close proximity to the walls(even closer if you extrapolate backwards from figure 1 to the point right after the 180 deg turn) and It seems like ceres scan matcher should be able to correct the error without the backend having to kick-in. I tried tuning most of the configuration values that seem relevant to real-time operation of cartographer(ceres scan matcher, voxler and motion filter) but I couldn't get any better results than what I have in the attached images. I m overlooking something here. Has anyone of you dealt with this problem before and know what could cause this? Thanks in advance.

[image]
**Figure 1:** Snapshot of the map as estimated by SPA. The robot was travelling as depicted by the yellow arrows at about 1 m/s. The green dots show the laser returns.

[image]
**Figure 2:** Snapshot of the map as estimated by SPA a few seconds later. The robot was travelling as depicted by the yellow arrows a about 1 m/s. The green dots show the laser returns.

PS: I am using cartographer with our in-house framework and I have a wrapper around the MapBuilder that conceptually replicates the ROS package. I will be attaching the Lua configuration table that I am using shortly.

-- Anurag

ajakhotia, 2017-02-07:
Below is the Lua configuration I am using.

[CartographerConfigurationTable.lua](https://github.com/googlecartographer/cartographer/files/758700/CartographerConfigurationTable.txt)

ajakhotia, 2017-02-07:
Below is the Lua configuration I am using.

[CartographerConfigurationTable.txt](https://github.com/googlecartographer/cartographer/files/758712/CartographerConfigurationTable.txt)

SirVer, 2017-02-08:
> I am using Cartographer for 2D slam on a robot with a 10m range, 270 deg FOV, 15 Hz planar laser > scanner and wheel encoders. It seems like ceres scan matcher is incorrectly estimating the yaw [...]

What you are describing is the local error accumulated by local slam which is then fixed by loop closure. Depending on our sensors, there is probably not much you can do about these (but a few suggestions below). If you are concerned about the jumping in the pose, you should probably use the odom transform that Cartographer also publishes which is always smooth.

> As you can see in the images, the scans are in a very close proximity to the walls(even closer if you extrapolate backwards from figure 1 to the point right after the 180 deg turn) and It seems like ceres scan matcher should be able to correct the error without the backend having to kick-in.

Your IMU readings (which probably will not integrate to 180 degrees perfectly) are fighting the scan matcher. You also try to SLAM at 4cm, which your laser will probably have a higher noise than that already - which makes it harder to scan match perfectly. And this allows for some wiggle room in matching walls which can easily end up being slight errors in the angle. 

> I tried tuning most of the configuration values that seem relevant to real-time operation of cartographer(ceres scan matcher, voxler and motion filter) but I couldn't get any better results than [...]

It seems to me that you are quite confident in your laser (given that you SLAM at 4cm). Have you tried disabling the IMU and only relying on the laser to not have to deal with this compromise? See https://github.com/googlecartographer/cartographer_ros/issues/249 where I suggested something similar for another user including the changes that I did to make this work. I would also try to crank up the max laser range to more than 10 m - it cold help to see farther away walls after the turn since they provide more information for fixing angles.

Your visualization looks gorgeous, btw. :)

closed 2017-02-08 by SirVer

ajakhotia, 2017-02-08:
I highly appreciate your quick and detailed response. Thank you. I am documenting my thoughts and effort below.

> Your IMU readings (which probably will not integrate to 180 degrees perfectly) are fighting the scan matcher. You also try to SLAM at 4cm, which your laser will probably have a higher noise than [...]

I am not using an IMU in my system but I am assuming that you meant Wheel Encoders instead of IMU. From my understanding of cartographer, correct me if I am wrong, the Wheel Encoder/IMU data are used to generate a seed position for the front-end ceres scan matcher which optimizes the scan matches to minimize  a combined score of scan match + delta shift from the seeded position + delta shift from the seeded orientation which are weighted based on the following three parameters from the lua configuration.

> TRAJECTORY_BUILDER_2D.ceres_scan_matcher.occupied_space_weight
> TRAJECTORY_BUILDER_2D.ceres_scan_matcher.translation_weight
> TRAJECTORY_BUILDER_2D.ceres_scan_matcher.rotation_weight

I did set the translation_weight and rotation_weight to something very very small in one my runs, which should, in effect, suppress any say of the odometry in the final optimal answer. However, I didn't see any improvement in the performance. I ll post some snapshots eventually.

>It seems to me that you are quite confident in your laser (given that you SLAM at 4cm). Have you tried disabling the IMU and only relying on the laser to not have to deal with this compromise? See googlecartographer/cartographer_ros#249 where I suggested something similar for another user including the changes that I did to make this work. I would also try to crank up the max laser range to more than 10 m - it cold help to see farther away walls after the turn since they provide more information for fixing angles.

The scanner I am using has standard deviation of about a centimeter. I did try running with just laser scans as the input, but the speed of the robot was way too much for the scan matcher to handle. That being said, I did not have the chance to tweak the motion filter to see if I can get a better solution. Will post an update if I do. The laser scanner maxes-out at 10 m range and doesn't give me any returns from beyond that range. 

> Your visualization looks gorgeous, btw. :)

Thank you! Your compliments have been conveyed.

wohe, 2017-02-10:
> The scanner I am using has standard deviation of about a centimeter. I did try running with just laser scans as the input, but the speed of the robot was way too much for the scan matcher to handle. 

Did you try `use_online_correlative_scan_matching = true`? Often times without IMU and odometry the Ceres scan matcher cannot find the correct rotation since it is too far from its initial estimate, and using the `real_time_correlative_scan_matcher` even with `linear_search_window = 0` works. A linear search window is really only needed if the constant velocity model is also not good enough as a starting point for the Ceres scan matcher.

ajakhotia, 2017-02-16:
Yes, I did try using correlative scan matcher but I ran into a different set of problems. I don't have enough computation power on the system to be able to run it a full scan rate.
    I started investigating into the ceres scan matcher in a standalone application using synthetic data. It's a bit early to make any claims, but from what I have observed, it seems like ceres scan matcher has a capture window of only 2.5~3.0 grid cells, i.e, for a 0.04 m grid cell size, I can get the scan matcher to converge to the right answer only if the seed is within 0.1 m range of the actual optimal position. This is under perfect conditions, where the synthetic scan I produce has zero noise and the probability grid has probability of 1.0 for the walls of a virtual room and 0.0 for the free space. Even with the clean data, the optimization is a hit or a miss. The scan matcher, sometimes, declares convergence too soon(with the criteria "Function tolerance reached") and produces a bad estimate(sometimes a diverged result as well) which I think is causing problems downstream. Also, this phenomenon is very sensitive to the occupied_space_weight parameter which makes me nervous.
    As I mentioned earlier, It's a bit early to make claims and I will soon post more information/evidence/code as I get further along my investigation.

And I apologize for my tardiness.

-- Anurag

wohe, 2017-02-17:
> It's a bit early to make any claims, but from what I have observed, it seems like ceres scan matcher has a capture window of only 2.5~3.0 grid cells, i.e, for a 0.04 m grid cell size, I can get the [...]

Your observation is correct and that is the intended behavior. The Ceres scan matcher uses local optimization over interpolated grid values, and is intended to fix small errors of up to one grid cell. If you have larger errors to correct, you have two options right now: use the correlative scan matcher, which is intended for correcting larger errors, or increase the size of the grid cells, which allows you to use the Ceres scan matcher for larger errors in meters.

Both options of course come with their set of advantages and disadvantages in memory consumption, processing speed, and precision.