cartographer-project/cartographer #201: getting started ?

phil123456, 2017-02-08:
Hello,

I read the quick start but there is not much to read

as I click on next, there are configuration variables, I have no idea which file should be opened to define these variables

also how do I add carographer to one of my program, how do I declare sensors/odometers and how do I feed cartographer with these data

what format should they have, how do I normalize these data ?

thanks

wohe, 2017-02-08:
Hi,

Did you follow our suggestion to try out the ROS integration? It gives you a good example how to use the Cartographer library should you decide not to use ROS and integrate directly with the library. I suggest getting the demos to run first: https://google-cartographer-ros.readthedocs.io/en/latest/

Regarding configuration variables, we provide example configurations for several use cases. Defaults are at https://github.com/googlecartographer/cartographer/tree/master/configuration_files and are overwritten, e.g., for the TurtleBot here: https://github.com/googlecartographer/cartographer_turtlebot/tree/master/cartographer_turtlebot/configuration_files

Please let me know if you have further questions.

Cheers,
Wolfgang

closed 2017-02-08 by wohe

phil123456, 2017-02-08:
still a bit vague, I am not planning (and I am not familiar) with ROS

I thought that cartographer was a c/c++ library

wohe, 2017-02-08:
It is a C++ library and you can use it without ROS. If you want to know how it is intended to be used, you should find it informative to look at how the ROS integration is implemented as an example.