cartographer-project/cartographer #202: Cartographer worse than Gmapping

nangong, 2017-02-09:
Launch file, Bag ,Configuration is [here](https://drive.google.com/open?id=0Bz4iO_gw1vXSX3lLYnZGdzBoLVE).

I tested the same bag file in gmapping and cartographer.  The two results are listed as follows,

Cartographer,
[image]

Just like I mentioned in #190 , the loop closure detection did not work.

Gmapping(which is very similar to the real environment),
[image]

Since I had already combined imu data into odom(/odometry/filtered in the ros bag), I did not provide the raw imu data(/costly_imu in the ros bag) to both slam.

Would you @SirVer  please tell what should I do next?

Thanks very much.

SirVer, 2017-02-09:
Your statement is that default settings are better than default settings of Cartographer? That seems unsurprising. What did you try to tune already? Have you tried disabling your IMU/odom merged topic and only use scan matching as I suggested in https://github.com/googlecartographer/cartographer_ros/issues/249?

nangong, 2017-02-10:
Thanks for replying so fast.

I tracked the odom trajectory and it was accurate in most time.

 After comparing the mapping progress between Gmapping and Cartographer, I found the problem was caused by what you alerted me in #190 ,i.e. my laser is not always level, it is fixed on the robot and the floor has tiny pitches in the following part,

[image]

Is that possible to fix this by tunning any parameters (e.g. increasing confidence of odom)?

SirVer, 2017-02-10:
> Is that possible to fix this by tunning any parameters (e.g. increasing confidence of odom)?

You can tune of course, but it is hard to say if this is fixable from just the image. If your sensors are strongly disagreeing, there will be no real way to fix the issue. If you trust odom so significanlty more than the laser and it has a bias (which it likely has), you will mess up other parts of the map.

I tried looking at your bag and configuration. But I do not have access to the drive elements. Also, make sure that your bag already contains the always-level laser changes.

nangong, 2017-02-10:
Sorry, I forgot to enable others to access the drive. It can be downloaded now.

alexbaucom17, 2017-02-28:
I am running into a similar problem where the loop closure is poor in indoor environments. I am using a laser, odom, and an IMU. The floor I am mapping also has small bumps and the robot does not always stay level. If I am providing IMU data should cartographer handle this okay? Or do I need to do some pre-processing on the laser data?

SirVer, 2017-03-01:
@alexbaucom17 It probably also just comes down to tuning. Sometimes IMU jumps over bumps are extreme and throw off the SLAM - then it might be better to disable IMU and use correlative scan (see closed issues labeled tuning for ideas.)

alexbaucom17, 2017-03-01:
Got it. I will do some more testing and tuning then. Thanks.

SirVer, 2017-03-02:
@nangong I found some time to look at this now. Unsorted comments:

- Your launch file refers to `demo_husky.lua` which you did not include. I used `demo_myrobot.lua` instead. 
- You also didn't include `demo_myrobot.rviz`. 
- Your configuration files have gone out of sync with HEAD, i.e. some options have vanished or have been renamed.
- You did not include a URDF and your bag contains TF information from another SLAM run - so I could not use that either. This means I cannot spatially relate your odom or IMU data to your laser data - I therefore disabled both odometry and IMU.

It would have been nicer to have a proper git fork with all files so I could reproduce on exactly your conditions - you it pretty hard for me to help you :(.

- Your outdoor (?) area that connects the two rooms the robot drives into seems to violate the 2D assumptions (i.e. flat floor, mostly orthogonal objects). Even to me as a human the laser clouds seem very different on the way back than they look on the way there.
- Sometimes your laser only sees very far away objects. Your configuration before threw away everything farther away than 30m, so there was no information for SLAM. 
- With your laser's poor angular resolution and high noise you cannot expect to SLAM at 5cm if your objects are so far away - your points visible jump by 15-30cm. I set the resolution to 15cm and that seemed to be sufficient to capture this.

My final settings drift by ~1.5m after the long loop (comping back to the first building) which is fixed by loop closure. Image attached below. I think adding IMU + odometry might increase performance (i.e. require less CPU) since you might get away without the correlative_scan_matcher, but this will probably  not improve quality.

The middle part of the map is still wrong. But I cannot blame SLAM there, the data is not looking consistent as mentioned above. This claim is supported that after loop closure both building parts are very good, but the middle part is still inconsistent - this is really only possible if the data is inconsistent there because everything is the same optimization problem.

My changes are [in this branch](https://github.com/SirVer/cartographer_ros/tree/user/nangong) in the last few commits.

[image]

closed 2017-03-02 by SirVer

nangong, 2017-03-03:
Got it..
Thanks very much for your help... : )

ahmdsamy, 2017-04-20:
@nangong  @SirVer I am using RPlidar which have a 6m range only .. and i am getting visual odometry from ZED camera, and the odom is very accurate.. how can i increase the odom confidence as you were saying in the upper comments ?

alexbaucom17, 2017-04-20:
@ahmdsamy They showed me how to change odometry confidence in [this issue](https://github.com/googlecartographer/cartographer_turtlebot/issues/48). You can look at the comments in the config file of my fork.