cartographer-project/cartographer #204: request ROS Indigo binary release

jack-oquin, 2017-02-15:
We are interested in running this package on multiple robots of the [UTexas Building-Wide Intelligence project](https://github.com/utexas-bwi). It would be very helpful to have a ROS Indigo release for cartographer which we could depend on.

If you are not familiar with the process of creating ROS binary releases, the [bloom package](http://wiki.ros.org/bloom) works great. After some [initial setup](http://wiki.ros.org/bloom/Tutorials/FirstTimeRelease), the job is quick and easy. 

I am willing to assist by answering questions, as are the other members of the [ROS discourse release category](https://discourse.ros.org/c/release). 

I believe other ROS users would also benefit from having a binary release available.

cc: @jsinapov, @Ashay-Lokhande

SirVer, 2017-04-24:
Cartographer team can't support maintaining a build for binary releases at this point in time. We would of course love to see such a thing happen though.

Paging @tfoote, can OSRF maybe help out here?

jihoonl, 2017-04-24:
ping @jspricke FYI

jack-oquin, 2017-04-25:
If you are willing to [make source releases](http://wiki.ros.org/bloom/Tutorials/PrepareUpstream), where appropriate, we would be willing to handle the ROS binary distribution steps.

damonkohler, 2017-04-26:
Thanks, Jack. That looks like a pretty low bar and I'm happy to take it on. I'll be out all next week, but I can look into it once I'm back.

closed 2017-05-18 by wohe

damonkohler, 2017-05-18:
Ok, @jack-oquin. I think we're all set for you to do the first binary release!

jack-oquin, 2017-07-04:
OK, sorry for slow response, I've gotten way behind on e-mail lately and still catching up.

jack-oquin, 2017-07-04:
Is there also interest in a Kinetic release?

damonkohler, 2017-07-05:
@jack-oquin Yes, I think so. We currently have CI configured for Indigo and Kinetic. We've had requests for Lunar, but aren't yet testing it.

PR #295 (2017-05-18): First unstable development release
Fixes #204