cartographer-project/cartographer #211: build error

sunmk2006, 2017-03-01:
when i catkin_make the cartographer_ros packet,it always print an error:

-- +++ processing catkin package: 'cartographer_ros_msgs'
-- ==> add_subdirectory(cartographer_ros-master/cartographer_ros_msgs)
-- Found required Ceres dependency: Eigen version 3.2.0 in /usr/include/eigen3
-- Found required Ceres dependency: Glog in /usr/include
-- Found Ceres version: 1.11.0 installed in: /usr/local
CMake Error at cartographer_ros-master/cartographer_ros_msgs/CMakeLists.txt:24 (include):
  include could not find load file:

    /functions.cmake

CMake Error at cartographer_ros-master/cartographer_ros_msgs/CMakeLists.txt:25 (google_initialize_cartographer_project):
  Unknown CMake command "google_initialize_cartographer_project".

-- Configuring incomplete, errors occurred!
See also "/home/viki/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/viki/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
i can't understand it,please help me.

wohe, 2017-03-01:
It seems CMake cannot find the `functions.cmake` file which is a part of `cartographer` not `cartographer_ros`. What confuses me a bit is that the error speaks of `cartographer_ros-master`. Did you follow the installation instructions at https://google-cartographer-ros.readthedocs.io/en/latest/? If not, what is your setup, e.g. are both `cartographer` and `cartographer_ros` in your catkin workspace and did you update both to the latest version?

sunmk2006, 2017-03-01:
Yes,you are right,i follow the installation instructions at( https://google-cartographer.readthedocs.io/en/latest/index.html#system-requirements),it did not have the latest version of cartographer. and now,i have update the cartographer,but another error:
CMake Error at /usr/local/share/cartographer/cmake/functions.cmake:102 (message):
  Compiling in debug mode is not supported.
Call Stack (most recent call first):
  cartographer_ros-master/cartographer_ros_msgs/CMakeLists.txt:25 (google_initialize_cartographer_project)

-- Configuring incomplete, errors occurred!
See also "/home/viki/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/viki/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

sunmk2006, 2017-03-01:
I've tried to follow ( https://google-cartographer-ros.readthedocs.io/en/latest/?) ,but it have an error:

==> Processing catkin package: 'cartographer_ros'
==> Building with env: '/home/viki/catkin_for/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/viki/catkin_for/build_isolated/cartographer_ros'
ninja: no work to do.
==> ninja -j4 -l4 in '/home/viki/catkin_for/build_isolated/cartographer_ros'
[9/16] Building CXX object CMakeFiles/...rtographer_ros/map_builder_bridge.cc.o
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -Wno-deprecated -O3 -DNDEBUG -isystem /usr/include/vtk-5.8 -isystem /usr/include/lua5.2 -isystem /usr/include/pcl-1.7 -isystem /usr/include/eigen3 -isystem /usr/include/ni -isystem /home/viki/catkin_for/install_isolated/include -isystem /opt/ros/indigo/include -I. -I/home/viki/catkin_for/src/cartographer_ros/cartographer_ros -I/usr/src/gmock/gtest/include -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/freetype2 -isystem /usr/include/libpng12     -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -pthread -std=c++11  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=uninitialized -O3 -DNDEBUG -MMD -MT CMakeFiles/cartographer_ros.dir/cartographer_ros/map_builder_bridge.cc.o -MF "CMakeFiles/cartographer_ros.dir/cartographer_ros/map_builder_bridge.cc.o.d" -o CMakeFiles/cartographer_ros.dir/cartographer_ros/map_builder_bridge.cc.o -c /home/viki/catkin_for/src/cartographer_ros/cartographer_ros/cartographer_ros/map_builder_bridge.cc
/home/viki/catkin_for/src/cartographer_ros/cartographer_ros/cartographer_ros/map_builder_bridge.cc: In member function ‘bool cartographer_ros::MapBuilderBridge::HandleSubmapQuery(cartographer_ros_msgs::SubmapQuery::Request&, cartographer_ros_msgs::SubmapQuery::Response&)’:
/home/viki/catkin_for/src/cartographer_ros/cartographer_ros/cartographer_ros/map_builder_bridge.cc:72:38: error: ‘cartographer_ros_msgs::SubmapQuery::Request’ has no member named ‘submap_index’
       request.trajectory_id, request.submap_index, &response_proto);
                                      ^
[9/16] Building CXX object CMakeFiles/...her_ros.dir/cartographer_ros/node.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer_ros': 
  Command '['/home/viki/catkin_for/install_isolated/env.sh', 'ninja', '-j4', '-l4']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/viki/catkin_for/build_isolated/cartographer_ros && /home/viki/catkin_for/install_isolated/env.sh ninja -j4 -l4

Command failed, exiting.

wohe, 2017-03-01:
Are you sure you have the unmodified latest source code? Did you do a clean build?

> error: ‘cartographer_ros_msgs::SubmapQuery::Request’ has no member named ‘submap_index’

This member exists since googlecartographer/cartographer_ros#151. What does your version of `SubmapQuery.srv` contain?

sunmk2006, 2017-03-02:
I'm sorry, I didn't clean up the old version,Now compiled successfully!Thank you very much!

closed 2017-03-02 by SirVer

ghy2013, 2018-07-16:
@sunmk2006  how to clean up the old version ,just to delete the file?

physics-lzs, 2019-03-17:
> I'm sorry, I didn't clean up the old version,Now compiled successfully!Thank you very much!

how to clean up the old version？I meet this question too, can you give me some help? Thank you very much!