cartographer-project/cartographer #212: why two lidars?

sunmk2006, 2017-03-01:
2D mapping only need one horizontal lidar,but your bag data always have another vertical_laser_2d,why?

wohe, 2017-03-01:
You're correct that 2D mapping only needs the horizontal LIDAR, and you can ignore the vertical LIDAR in this case.

Regarding the vertical LIDAR: Why not? We collected the data, so we put it in the bags. One use case would be to reconstruct geometry for which the horizontal LIDAR is not suitable since it mostly covers a single plane, if you ignore the fact that the backpack is not always perfectly level. Of course you are free to find other uses.

sunmk2006, 2017-03-01:
Did you have use the vertical lidar data in the 2D mapping process?Can be done with 2D laser reconstruction？

wohe, 2017-03-01:
I'm not sure I understand your questions. To run 2D SLAM, you do not need the vertical LIDAR, e.g. running
```
roslaunch cartographer_ros offline_backpack_2d.launch bag_filenames:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag
```
does not make use of it. You can make use of it though, e.g. using the SLAM result from above and running
```
roslaunch cartographer_ros assets_writer_backpack_2d.launch bag_filenames:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag trajectory_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag.pb
```
computes X-ray images using both LIDARs and stores the results at `~/.ros/xray_*` by default.

sunmk2006, 2017-03-02:
3D mapping you have tow lidars, Can I use a tilted laser to mapping,(eg. A  Velodyne VLP-16 lidar with a tilt of 45 degrees)to get trajectory and reconstruction?

SirVer, 2017-03-02:
​​

>
> 3D mapping you have tow lidars, Can I use a tilted laser to mapping,(eg. A
> Velodyne VLP-16 lidar with a tilt of 45 degrees)to get trajectory and
> reconstruction?
​​Answered here:
https://github.com/googlecartographer/cartographer/issues/213

—

Google Germany GmbH
Erika-Mann-Straße 33
80331 München

Registergericht und -nummer: Hamburg, HRB 86891
Sitz der Gesellschaft: Hamburg
Geschäftsführer: Matthew Scott Sucherman, Paul Terence Manicle

closed 2017-03-20 by SirVer