cartographer-project/cartographer #213: Can I use a Velodyne VLP-16 lidar for 3D reconstruction？

sunmk2006, 2017-03-02:
3D mapping you have tow lidars,a horizontal_laser_3d can do it too,but a vertical_laser_3d always failed, Can I use a tilted laser to mapping,(eg. A Velodyne VLP-16 lidar with a tilt of 45 degrees)to get trajectory and reconstruction?

SirVer, 2017-03-02:
Theoretically yes, but practically the opening angle of the VLP16 is pretty small. This means that you see little of the world in one revolution and rarely enough to constrain the point cloud in all dimensions. If your environment is feature rich though, it might just work.

closed 2017-03-02 by SirVer

yxting1994628, 2017-03-28:
Hello!Can you run cartographer with VLP16?

damonkohler, 2017-03-28:
Yes, but you'll need 2 and an IMU for it to work out of the box.