cartographer-project/cartographer #214: location perform not well

gongyue666, 2017-03-02:
hi，firstly thanks for your work.When I use the package with my robot I face with a problem that the location information from cartographer seem wave largely ,and sometimes I even suspect the (x,y) is reversal,which is (y,x)!! I use only an URG-10LX for the sensor,the laser works well with gmapping package.
here is my .lua format:`include "map_builder.lua"
options = {
  map_builder = MAP_BUILDER,
  sensor_bridge = {
    horizontal_laser_min_range = 0.3,
    horizontal_laser_max_range = 5.6,
    horizontal_laser_missing_echo_ray_length = 1.2,
    constant_odometry_translational_variance = 0.,
    constant_odometry_rotational_variance = 0.,
  },
  map_frame = "map",
  tracking_frame = "laser",
  published_frame = "laser",
  odom_frame = "odom",
  provide_odom_frame = true,
  use_odometry_data = false,
  use_horizontal_laser = true,
  use_horizontal_multi_echo_laser = false,
  num_lasers_3d = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}
MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
SPARSE_POSE_GRAPH.optimization_problem.huber_scale = 1e2
return options`

SirVer, 2017-03-02:
This Lua configuration is invalid. `sensor_bridge` is not an entry in the options struct, so this should not even start at HEAD. Did you try pulling recently?

>    horizontal_laser_max_range = 5.6

This seems awfully specific. Did you try increasing this?

Without your full configuration (urdf, launch files) and sample data (bag with laser scans) it is impossible to know what is wrong. Please provide a fork of `cartographer_ros` and add your changed/added configuration files there. Then provide a google drive link with bag data.

We did SLAM over sensor data from this particular laser before successfully and it worked well. The strong errors you describe let me think that you use a wrong URDF

gongyue666, 2017-03-03:
I am so sorry,I find out the bug ,it is mine .I publish the relation between laser and base_link in another node,so the location performed badly.Now it run very well!!!Thanks so much for this great open source~~

SirVer, 2017-03-03:
Good that you found it! Thanks for the kind words.

closed 2017-03-03 by SirVer