cartographer-project/cartographer #215: trajectory builder - outdoor data

aksgoel, 2017-03-02:
I am trying to build a 3d map from a rosbag

I serve data needed by cartographer_ros by doing the following things on our rosbag:
Republish our IMU data by inverting the direction of Z acceleration. I also set the angular velocity co-variance to 0 for all fields.
Use VLP16 driver and republish velodyne_points as points2. We publish one point cloud per velodyne packet.
Publish static_transforms between base-link and imu and base-link and velodyne.

The challenge I am facing is our trajectory builder isn’t building more than one trajectory. I.e. I am facing trouble with the pose estimation that is published as the map->odom transform by cartographer. I haven’t changed any parameters in trajectory builder or sparse_pose from the main cartographer source code. 

All the matched scans overlay on each other when visualized in rviz.

I think that there is some configuration there that needs modification because the velocity of the robot is much greater than the velocity of the backpack.

Thanks for your help. Let me know if you need more information

We are using the demo_car_3d.launch

Here is the ros source code: https://drive.google.com/open?id=0BzptGYp7_SKtY3JmaE1yQURsOXM

The ROSBAG we are using: https://drive.google.com/open?id=0BzptGYp7_SKtLUtrR3dZZzY3TGs

SirVer, 2017-03-03:
> The challenge I am facing is our trajectory builder isn’t building more than one trajectory. 

I do not understand this sentence. Multi-trajectory is a term we use if more than one run (i.e. two or more bags usually) are replayed into the same Cartographer node and loop closing is applied to all trajectories together. I do not think that is what you are trying to do?

aksgoel, 2017-03-05:
Apologize for terminology mismatch: we are not yet trying to replay multiple rosbags, to apply loop closing. We are trying to build trajectory for first pass - however the pose estimated for the first pass is incorrect.

Also curious if a second LIDAR or Odometry info will fix this. (We are using IMU and one Velodyne 16 at the moment).

SirVer, 2017-03-20:
@aksgoel I do not have access to your bag it seems.

aksgoel, 2017-03-21:
sorry, please try now?

SirVer, 2017-04-24:
I finally got around at looking at your data. I have a hard time making sense of what I see there:

- I started with `cartographer_ros` master and copied car_3d.lua, demo_car_3d.rviz, car_3d.launch, demo_car_3d.launch, car_3d.urdf, demo_car_3d.launch, car_3d.urdf. I also downloaded your bag.
- Your bag contains two IMU topics: /imu/data which gets remapped to /imu_old. You mentioned that you republish the IMU, but I can see neither a node/configuration for doing so in your launch files nor remapped data in the bag. So I used the other available IMU topic: `/vehicle/imu/data_raw`. This one is broken:
    - it sets `angular_velocity_covariance[0] = -1` and the same for linear_acceleration_covariance. [According to ROS docs](http://docs.ros.org/api/sensor_msgs/html/msg/Imu.html) this means that you have not measured the values - but you clearly have. Instead the covariance should be all zeros if you do not know it, but have measured data. Cartographer actually checks for this and terminates if the data states to not contain angular velocities or linear accelerations. I disabled the checks for experimenting with your data. 
    - the `frame_id` of the IMU data is the empty string. I patched it to be `imu`.
- The `/velodyne_packets` only comen in at around 10 Hz. I would expect way more if this was every single UDP packet the Velodyne sends. I rather think that the `velodyne` nodes collects one full rotation of UDP packets before sending on this topic. The created `/velodyne_points` are also 10 Hz and look in rViz to be full revolutions, so this supports my theory. Full revolutions are [suboptimal](https://google-cartographer-ros.readthedocs.io/en/latest/faq.html#the-laser-data-in-the-3d-bags-is-much-higher-than-the-maximum-reported-20-hz-rotation-speed-that-the-vlp-16-can-do-why).
- Your data collection has gaps of several seconds (see rqt_baq screenshot) and the IMU and the laser time stamps seem unrelated.
- The velodyne data alone also seems wonky. I cloned the [velodyne ROS](https://github.com/ros-drivers/velodyne) repository so I can use your launch file. I assume you did not modify the velodyne driver locally? Looking at the output of the Velodyne in its own frame in rViz it looks like your car is permanently in a bowl (see 2 screenshots from rViz below). This is clearly wrong as you can see other cars passing yours and they drive along the 'bowl' in rViz, though the road must be flat in the real world.

In summary I feel that your data is broken - I was unable to feed it sensibly to Cartographer. I will close this issue until you can provide cleaner data. Sorry again for taking so long to respond.

rViz Screenshots showing the Velodyne inside a bowl:

[image]
[image]

rqt_baq screenshot, showing gaps in data:
[image]

SirVer, 2017-04-24:
My experiments are in this branch: https://github.com/SirVer/cartographer_ros/tree/user/askgoel

closed 2017-04-24 by SirVer