cartographer-project/cartographer #217: how to build point cloud viewer?

sunmk2006, 2017-03-03:
When I did the third step :“npm install", there was an error: npm WARN point_viewer@0.1.0 No repository field.

SirVer, 2017-03-03:
Did you work it out?

sunmk2006, 2017-03-04:
build successfully, but：
viki@viki:~/soft/point_cloud_viewer-master/target/release$ ls
build  build_octree  deps  examples  libpoint_viewer.rlib  native  viewer
viki@viki:~/soft/point_cloud_viewer-master/target/release$ build_octree
build_octree: command not found
viki@viki:~/soft/point_cloud_viewer-master/target/release$

sunmk2006, 2017-03-04:
./build_octree generates a lot of files,but
./viewer /home/viki/source/data/cartographer_data/mode
Listening on port 5433.
IT seems to do nothing, No changes in memory and CPU ，What happened？

SirVer, 2017-03-06:
open your browser to `http://localhost:5433`, then left click and drag a bit. It should load your points. use WASD to fly around, mouse wheel to adjust speed.

closed 2017-03-06 by SirVer

sunmk2006, 2017-03-06:
You are right!!!