cartographer-project/cartographer #239: Tools for integration testing

macmason, 2017-04-25:
(This might end up in `cartographer_ros`.)

We need tools for doing integration testing: given some data, confirming that Cartographer's behavior doesn't change over time.

This is intended as a tracking issue for work TRI's planning to do on this.

@damonkohler @SirVer @wohe once upon a time, there was code for doing trajectory comparisons. Could that be made available?

@yutakaoka for TRI.

wohe, 2017-06-16:
I just fixed #310. The idea is to use this generated ground truth data as input to `compute_relations_metric_main`, i.e.
1. Extract the computation of errors into a separate file.
2. Add support for reading a `GroundTruth` proto in addition to a text file with unix timestamps.
3. Use it for an integration test.

SirVer, 2017-11-15:
@cschuet Is this done with the nighly pipeline running?

cschuet, 2017-11-23:
With the inclusion of ground_truth data in the nightly pipeline I call this done.

closed 2017-11-23 by cschuet