cartographer-project/cartographer #244: Support tracking external observations (Landmarks) as constraints

SirVer, 2017-05-02:
Users expressed interest to add constraints to visual features, GPS tracks, or bluetooth beacons as constraint for the SLAM. This is the tracking issue for discussion and implementation of this feature.

ahundt, 2017-06-08:
Here is an example of a mobile base with an AR tag on the gripper, as I had mentioned on today's google hangout:

[image]

In this case the landmark may move relative to both the sensor and world. Both sensors and grippers could be on the robot, the gripper, or the wall. Furthermore, with an AR tag on the base in addition to the gripper poses based on individual joint angles could be estimated and incorporated to refine the estimate. 

I understand this case may be out of scope but even if it cannot be incorporated, it could still useful to consider that landmarks such as AR tags may move in known manner relative to both the world and the robot. Perhaps the API design might then be set up such that it can be more easily extended for this sort of data later on.

jihoonl, 2017-06-09:
@SirVer mentioned that the currently proposed API in #246 is not sufficient in some use cases during the open house meeting. Could you explain what is missing?