cartographer-project/cartographer #253: Add full serialization of Cartographer's state

SirVer, 2017-05-03:
This can't be a single protobuf, since this would require a ton of memory.

Instead the approach should be to invent a container proto, with a set of [`oneof`](https://developers.google.com/protocol-buffers/docs/proto3#oneof)s for each individual piece of state and streaming it through a [RecordWriter](https://github.com/wohe/cartographer/blob/98fffc0e3b15f7b7669917e10be16d54c1a49d1d/cartographer/io/recordio.h).

SirVer, 2017-05-03:
/cc @brandon-northcut Would you give this a shot?

brandon-northcutt, 2017-05-04:
@SirVer Yes. I'm working on it now. I'll make a PR once I have something I think is going in the right direction. Speaking of, what is the preferred workflow for a PR?

SirVer, 2017-05-04:
> Speaking of, what is the preferred workflow for a PR?

We are not particularly picky. Just open one whenever you want a first look or feel it is ready for merging, then rebase or merge master regularly. Once we merge, we will flatten the whole PR into a single commit on master.

wohe, 2017-05-18:
@brandon-northcutt Any update? Please let me know if there is anything blocking you I can help with.

wohe, 2017-06-06:
Friendly ping. @macmason What's the state of this?

macmason, 2017-06-06:
I believe this is still in @brandon-northcutt 's queue.

Brandon, let me know if that's wrong.

brandon-northcutt, 2017-06-06:
@wohe @macmason : Sorry for the delay. I'm hoping to get this PR in before the round-table Thursday so we can discuss it.

brandon-northcutt, 2017-06-13:
Looks like trajectory.proto doesn't include range data. Is it true that each trajectory node has either a 2D range scan or a 3D compressed range scan?

SirVer, 2017-06-14:
@brandon-northcutt Yes.

jihoonl, 2017-07-04:
@wohe you mentioned that you want to serialize odometry in [map_builder.cc:157](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/map_builder.cc#L157). Is it necessary? Isn't odometry used only for local trajectory building?

wohe, 2017-07-04:
Yes to both. Right now it is not included in the pose graph optimization, but it should be.

SirVer, 2017-07-05:
We serialize submaps and the sparse pose graph now. Still missing is range data, IMU and odometry.

brandon-northcutt, 2017-07-06:
Range data is almost done in #334 and it looks like the IMU sensor proto landed in #381. I'll take a crack at odometry if no one else is working on it. We can discuss at the round table tomorrow. I'm excited to do some experiments with full serialization and deserialization carrying on from #379.