cartographer-project/cartographer #254: Change the fixed point of Cartographer's map away from being the initial pose of the first submap

SirVer, 2017-05-03:
/cc @dcconner @brandon-northcutt

This was pulled out from https://github.com/googlecartographer/cartographer/issues/197 to focus discussion. 

In https://github.com/googlecartographer/cartographer/issues/197#issuecomment-298477329 the issue was raised that the robot position jumps whenever loop closure occurs. The suggestion was to update prior submap poses relative to the current robot pose estimate.

ROS solves this differently and Cartographer follows this convention. This is documented [in REP 105](http://www.ros.org/reps/rep-0105.html). The local pose is mapped in a different global frame usually called `odom` and Cartographer can publish this. `odom -> base_link` never jumps and can therefore be used as local pose. `map -> odom` always changes when loop closure happens (and jumps) to keep the robots pose correct in the current world frame.

This is usually sufficient to express the needs of algorithms and observers. My inclination is therefore to close this issue as `wontfix`. Am I missing something here?

Entropy512, 2017-05-03:
At least from my perspective, the benefit of having the origin of the map somewhere else other than the initial pose of the first robot to begin mapping is less of a benefit for robot navigation and more of a potential human interface/UI perspective if Cartographer is integrated into a higher-level navigation solution that presents a representation of a map to a human - there's likely a strong desire to have the origin point of the map at a location known to be the center (or a corner) of a facility.  There's an even stronger desire to have the x/y coordinate system axes be aligned with the typical horizontal grid structure of most facilities.

One solution to the latter is simply to ensure that the first robot to start mapping the facility starts aligned along the desired axis.  I've done this at least once - but my robot's typical parking spot is at an odd angle that results in a human-unfriendly coordinate system.

Another possibility might be to just publish a transform between "map" and a "human_friendly_map" coordinate system - but then the challenge becomes deciding and designating which submap is the anchor point for this transform.

Currently it's not a big issue since you can just rotate the map image written out by the finish_trajectory service, but once you start doing persistent mapping there could be some UI issues if the map coordinate system's origin isn't easily "anchorable" to a reference point that's human-friendly, whether by choosing an alternate origin coordinate for map, or by publishing a transform between map and that particular anchor point.

SirVer, 2017-05-04:
> Another possibility might be to just publish a transform between "map" and a "human_friendly_map" coordinate system - but then the challenge becomes deciding and designating which submap is the anchor point for this transform.

This is the ROS approach and seems the most flexible - adding this feature into Cartographer will duplicated ROS functionality and it will also add some bias on which sort of transform people want. Doing this outside of Cartographer seems the correct approach to me. 

> Edit: Some of this might be relevant to #244 since those landmarks are going to be in a predefined coordinate system that does not have the origin at the first robot pose.

Landmarks are observations and therefore relative to the trajectory of the observing robot - at least initially.

> We might also want to enhance the documentation to make this behavior clear to users. (It might be there and I just missed it.)

Probably not. Our documentation is pretty thing. This would be a great opportunity for helping out.

dcconner, 2017-05-04:
With regard to #244, if the landmarks are at a known position (e.g. GPS fixes or known markers), then they are not just relative observations tied to a trajectory.  If #244 was implemented with external landmarks at known locations, then this issue of defining a different fixed point might become relevant; however, given other discussions, it does not seem to be a priority now.

SirVer, 2017-05-04:
Sorry, I was unclear in my explanation: Map-fixed landmarks can be approximated as pose-relative landmarks - for example by tying them to the origin and "observing" them at time = 0. Therefore we have no plans to support them any differently as relative ones. 

GPS is more complex, we are thinking about this right now in https://github.com/googlecartographer/cartographer/pull/246. I think we underestimated the complexity of possible landmark observations in this simple example.

Entropy512, 2017-05-05:
Ah, I think I understand what you're getting at - even if the map-fixed landmark isn't actually observable from the sensor platform at time 0 (when Cartographer is started), you provide an "observation" of the sensor at its known fixed position.  (Hmm, although wouldn't this require knowing the sensor platform's precise location at time 0, or the ability to redefine the sensor platform's position at time 0 as being something other than (0,0,0)?)

And for figuring out the transform between map and (for example) a human_friendly_map coordinate system - one could use GetSubmapTransforms()?  (Mofef's PR caused a light bulb to go off in my head)

SirVer, 2017-05-05:
> (Hmm, although wouldn't this require knowing the sensor platform's precise location at time 0, or the ability to redefine the sensor platform's position at time 0 as being something other than (0,0,0)?)

I do not think so: the observations are free to float in the same way than the submaps themselves. Only the origin stays put and everything is relative, so this should work.

> And for figuring out the transform between map and (for example) a human_friendly_map coordinate system - one could use GetSubmapTransforms()? 

I do not understand the question. You can define this transform to whatever you feel like. GetSubmapTransforms gives you the position of the submaps relative to the origin.

Entropy512, 2017-05-22:
> I do not think so: the observations are free to float in the same way than the submaps themselves. Only the origin stays put and everything is relative, so this should work.
Hmm...  I'll need to look in more detail at how things are structured.  I'm probably misunderstanding you, but it sounds like your intent is to use a landmark solely to handle relevant movement of the robot between observations of that landmark?  What about when you have two landmarks whose position relative to each other is known precisely?  (this starts to approach the problem of how to handle something like GPS).

Also one thing to keep in mind is that for some applications (UWB location beacons such as the various solutions using Decawave's DWM1000), observations with range but not any form of angle might be available.  (I see this as being potentially useful in a long aisle/hallway with insufficient features to determine distance down the hallway, in the case where vehicle odometry is thrown off by a skid caused by a hard braking event.)  In fact now that I think about it, this is almost identical to the problem of using GPS as an observation (assuming, for GPS, that you have access to the raw pseudorange data).

> I do not understand the question. You can define this transform to whatever you feel like. GetSubmapTransforms gives you the position of the submaps relative to the origin.
Yes - and I think that for someone who wants to have the origin of a "human friendly map" be within a non-origin submap, GetSubmapTransforms is exactly what someone would need to determine the transform between their desired anchor point and the origin for Cartographer's "map" reference.

An example:  Suppose that someone wants the origin to their map be a column in the center of a warehouse.  At some point, a submap will contain scans of that column.  Someone could define that submap as the origin of the coordinate system they would prefer to work in, and use GetSubmapTransforms() to determine the transform needed between Cartographer's coordinate system and theirs.  (A fixed transform wouldn't be suitable, as it may change after an optimization run.)

All of this would happen outside of Cartographer - but with GetSubmapTransforms() I think Cartographer has all of the hooks necessary for someone to do this.

SirVer, 2017-05-26:
No, I think you understood me well. We do not plan to handle landmark <> landmark constraints at this point in time since they are quite a different cup-of-tea from the construction of the optimization problem. For now, Cartographer is all pose-relative, so this is the only thing we want to support for the moment.

> Also one thing to keep in mind is that for some applications (UWB location beacons such as the various solutions using Decawave's DWM1000), observations with range but not any form of angle might be available.

Yes, we are aware of this, but we did not model it in our landmarks API yet. We plan a revision of the API, because this seems to be a use case we somehow have to model.

SirVer, 2017-07-24:
Closing since there is agreement that this can be modeled outside of Cartographer without loss of generality.

closed 2017-07-24 by SirVer