cartographer-project/cartographer #255: Support starting against a big "composite" map loaded from disk

SirVer, 2017-05-03:
This was pulled out from https://github.com/googlecartographer/cartographer/issues/197. The idea was first suggested in https://github.com/googlecartographer/cartographer/issues/197#issuecomment-298477329.

Original question: Would it be possible to save and load a single "composite" map as a global submap, and then use that to initialize optimization?

The answer is no, not easily. Cartographer uses Submaps to do granular matches against and allowing to float the global constraints around. Having one giant submap as initialization will probably be fairly inefficient and for loop closing this against other maps we would also require the laser scans that belong to this anyways.

So to support this would be a lot of work and introducing new paradigms into Cartographer. It also requires more state than just the map. And if it works, it will give inferior results to true lifelong mapping. So my inclination is to close this as `wontfix`. Thoughts?

SirVer, 2017-05-03:
/cc @dcconner

dcconner, 2017-05-04:
No objections to closing.  Cartographer has a different focus.

closed 2017-05-04 by SirVer