cartographer-project/cartographer #260: /usr/bin/ld: cannot open output file cartographer/io/proto_stream_test: No such file or directory

jspricke, 2017-05-04:
Hi,

trying to compile current master (12b3cc0) on Debian unstable with `make -j12` gives:
```
/usr/bin/cmake -E cmake_link_script CMakeFiles/cartographer/io/proto_stream_test.dir/link.txt --verbose=!
/usr/lib/ccache/c++   -Wall -Wextra -Wpedantic -Wabi -Wconversion -pipe -fdiagnostics-color=always -O3 -Weffc++ -Wold-style-cast -O3 -DNDEBUG  -Wl,--as-needed -s CMakeFiles/cartographer/io/proto_stream_test.dir/cartographer/io/proto_stream_test.cc.o  -o cartographer/io/proto_stream_test -rdynamic libcartographer.a gmock/libgmock_main.a -lpthread /usr/lib/libceres.so.1.12.0 -lglog /usr/lib/x86_64-linux-gnu/libgflags.so.2.1.2 -llua5.3 -lm -lboost_iostreams -lboost_regex -lwebp -lglog -lgflags -lcairo -lprotobuf -lpthread -lpthread
/usr/bin/ld: cannot open output file cartographer/io/proto_stream_test: No such file or directory
```
`mkdir cartographer/io/` solves the problem, but I've no idea why the directory is not created.

jspricke, 2017-05-04:
It start's breaking in 6d2feaa8c7faf6b91f6616945b91c75785b18f39

SirVer, 2017-05-04:
@jspricke Please run `rm -rf *_isolated` and try again. We renamed all tests through the new globbing and catkin is not clever enough to understand this :(.

Closing, because this should fix it.

jspricke, 2017-05-04:
I did so, it did not ;). Please reopen.

jspricke, 2017-05-04:
I don't use catkin but plain cmake.

jspricke, 2017-05-04:
Just for completeness, I tired `catkin_make_isolated` in a clean workspace and it's failing as well. @SirVer please reopen.

wohe, 2017-05-04:
@jspricke Are you certain that it did not start breaking at #241 when the new test was added? I'm asking because I cannot see anything special about this particular test in #258. How did you come to the conclusion that #258 broke you?

jspricke, 2017-05-04:
I cherry picked 12b3cc0 on top of ff06e37 and it worked and master does not.

SirVer, 2017-05-05:
@jspricke This confuses me. We build on debian stretch on Travis and it does pass. Could you repeat step by step what our CI does (just follow the [Dockerfile.stretch](https://github.com/googlecartographer/cartographer/blob/master/Dockerfile.stretch)) and see if that fails for you too? I am looking for the discriminating factor that is different for you than for Travis.

jspricke, 2017-05-05:
I've just tried it again a couple of times and can't reproduce it anymore :(. Closing as I've no idea what could have produces it. Sorry for the noise.

closed 2017-05-05 by jspricke