cartographer-project/cartographer #264: Cell limits should be size_t instead of int

catskul, 2017-05-05:
in `xy_index.h` and hence `cell_limits.proto`,  `num_x_cells` and similar should be `size_t` or `uint` and rather than the current `int` and `int32`

While it might be unusual to think of needing anywhere near max int number of cells in a particular direction, c++ promotion rules mean that for instance when calculating the total number of cells with  greater than sqrt(numeric_limits<int>) + 1 in each direction, incorrect overflow behavior will happen unless the user static casts each to size_t first.

```
#include <limits>
#include <cstddef>
#include <iostream>
#include <cmath>

using namespace std;

auto main() -> int {
    int num_x_cells = sqrt(numeric_limits<int>::max()) + 1;
    int num_y_cells = sqrt(numeric_limits<int>::max()) + 1;
    int total_cells_int       = num_x_cells * num_y_cells;
    size_t total_cells_nocast = num_x_cells * num_y_cells;
    size_t total_cells_wcast  = static_cast<size_t>(num_x_cells) * static_cast<size_t>(num_y_cells);

    cout << "num_x_cells:   " << num_x_cells << "\n";
    cout << "num_y_cells:   " << num_y_cells << "\n";
    cout << "total_cells_int:    " << total_cells_int    << "\n";
    cout << "total_cells_nocast: " << total_cells_nocast << "\n";
    cout << "total_cells_wcast:  " << total_cells_wcast  << "\n";

    return 0;
}
```

results in:

```
num_x_cells:   46341
num_y_cells:   46341
total_cells_int:    -2147479015
total_cells_nocast: 18446744071562072601
total_cells_wcast:  2147488281
```

I propose they be changed to `size_t` and the equivalent for the proto file.

Not sure if there's likely to be knock on changes necessary.

SirVer, 2017-05-08:
The reason it is an integer instead of a size_t is outlined in the [Google C++ style guide](https://google.github.io/styleguide/cppguide.html#Integer_Types), in the sub section "On Unsigned Integers".

Your argument is of course valid, but since the `cell_limits` are intended for `HybridGrid`s/`Submap`s,  `max<int>() - 1` with a resolution of 5cm per grid cell is still > 2km - way larger than any submap should probably ever be. My inclination is to say that the dangers outlined in the style guide are more valid than the dangerous outlined by you in this case - so I'd say `int` is more correct.

Is there some background to your bug report? Did you run into issues with the current data type?

catskul, 2017-05-08:
I was reading some code and noticed `static_casts` all over the place for otherwise simple calculations. In investigating why it was necessary, I noticed the ints in CellLimits.

The report is mostly an issue of avoiding compiler warnings and/or noisy static casts all over the place.

But, I accept the points made by the google style guide.

Separately I might argue (perhaps in separate ticket at some point in the future) that 2km is on the same order of magnitude of what might be necessary, esp with regard to airports or tunnel connected sets of buildings like university campuses, theme parks, etc, and so may be worth upping to an `long int`/`int64`.

https://en.wikipedia.org/wiki/List_of_longest_buildings

SirVer, 2017-05-09:
The 2km argument is only for a single Submap (a single HybridGrid), Cartographer already scales way higher - we mapped cities with it. You will have hundreds of submaps by then.

closed 2017-05-09 by SirVer