cartographer-project/cartographer #265: Build issues introdu

ojura, 2017-05-08:
Hello, a recent commit (6d2feaa8c7faf6b91f6616945b91c75785b18f39) by @wohe has broken building libcartographer (this repository) for me.

When running CMake, the first sign that something is wrong are loads of warnings about target names, like the following:
```
CMake Warning (dev) at cmake/functions.cmake:36 (add_executable):
 Policy CMP0037 is not set: Target names should not be reserved and should
  match a validity pattern.  Run "cmake --help-policy CMP0037" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  The target name "cartographer/common/blocking_queue_test" is reserved or
  not valid for certain CMake features, such as generator expressions, and
  may result in undefined behavior.
```
The full CMake log is here: https://gist.githubusercontent.com/ojura/da7e604a5f0c9a321c9bf6259c4e464f/raw/e880dd6b9fc6c064540cb2bd3bb82d8b79715bdc/cmake_warnings

Also, I get a linking error with one of the tests, here's the build log: https://gist.githubusercontent.com/ojura/da7e604a5f0c9a321c9bf6259c4e464f/raw/e880dd6b9fc6c064540cb2bd3bb82d8b79715bdc/build_error

I think you might have missed this because this doesn't happen when Cartographed was previously successfully built. Please try deleting your CMake build directories and try building Cartographed; I believe you'll get the same linking problem. 

Doing `git revert 6d2feaa8c7faf6b91f6616945b91c75785b18f39` allows me to built Cartographer again.

wohe, 2017-05-08:
Can you try a clean build? It seems to me that your logs list the old test names. My understanding is that this happens because the tests were renamed, and CMake (at least how it is used in Cartographer) is not robust to this kind of change to the CMakeLists.txt.

I will look into fixing the CMP0037 warning. I was not aware of this restrictions of target names in more recent CMake version. 2.8.12.2 did not complain.

ojura, 2017-05-08:
I _have_ tried a clean build (`mkdir build; cmake ..; make -j8` is as clean as it gets); in fact, I mentioned above (semifinal paragraph) that it happens indeed when you try a clean build, and doesn't if you already compiled an earlier commit of Cartographer :) It seems to me that somewhere you still link the old tests, and you haven't noticed this because they are lingering in your build tree from a previous build and hence the linking succeeds for you.

ojura, 2017-05-08:
If my explanation above is correct, maybe it would be a good idea to completely clean the build tree in Travis to catch errors like this one.

Speaking of that, would you mind looking into one more build issue, although it's not very criticial - my builds on travis have for a long time been occasionally failing like this: https://travis-ci.org/ojura/cartographer_ros/builds/224350625
It complains about not being able to load the cache. This can usually be fixed by going into Travis and deleting the cache manually using the web interface.

wohe, 2017-05-08:
Not yet clear to me what is happening. We always do clean builds in our CI, i.e. if a PR gets merged it passed all unit tests after a clean build.

You're not the first one to run into this issue though. What confuses me is that the test that fails to build seems to be because of a non-existing directory. That test was recently added in PR #241. I'm not sure why your build would not have failed then already.

1. What Linux are you using?
2. Does it also fail with ninja?
3. Did you install libcartographer system wide? If you did: Does it still fail if you uninstall first?

ojura, 2017-05-08:
1) Ubuntu 16.04.2
2) Oddly, no!
3) No.

I confirmed that you can reproduce the problem by booting up a live Ubuntu 16.04.2 live CD and executing the following:

```
#!/bin/bash
sudo add-apt-repository universe
sudo apt update
sudo apt install git -y
git clone https://github.com/googlecartographer/cartographer.git
git clone https://ceres-solver.googlesource.com/ceres-solver
./cartographer/scripts/install_debs.sh
cd ceres-solver
mkdir build
cd build
cmake ..
make -j8
sudo make install
cd ../..
cd cartographer
mkdir build
cd build
cmake ..
make -j8
```

wohe, 2017-05-08:
Okay. Good to see that it now failed with a different test. This makes it indeed more likely to be related to #258. No idea why make does not like it, but now that tests got renamed again and no longer contain slashes in their name: @ojura Can you check whether #267 fixed the issue? In case this was problem.

ojura, 2017-05-08:
I just checked, #267 fixed it! It seems that CMake + make really hates slashes in target names. Great, thanks! If you find the time, please have a look at that Travis issue as well.

closed 2017-05-08 by ojura

wohe, 2017-05-09:
Regarding the docker cache issue: If you only had this Travis problem once, I'd suggest just clearing the Travis cache. Be sure to let us know if it happens more often.

ojura, 2017-05-09:
Okay. I've had it happen multiple times in the past few months though, and had to clear the Travis cache a few times to get it to pass. When I stumble into it again, I will let you know.