cartographer-project/cartographer #292: Travis issues

ojura, 2017-05-16:
@wohe told me to get in touch if I run into Travis cache issues again. Have a look here: https://travis-ci.org/larics/cartographer_ros

Let me know if I can do something to help debugging this.

ojura, 2017-05-18:
Here's some more details.

 `DOCKER_CACHE_FILE=/home/travis/docker/trusty-cache.tar.gz`.

 When this happens, this file is 20 bytes long, and contains the following bytes (base64 encoded): `H4sIADj6GlkAAwMAAAAAAAAAAAA=`. It's a .gz file which unzips to zero bytes. It seems that `docker load` doesn't like when it's fed a zero-byte file, and it throws an "invalid argument", along with some non-zero status code, which causes the whole build to exit because `errexit` is set.

It seems that the options are either to disable the `errexit` mode, or to unzip the file to e.g. /tmp/, check its size for > 0, and then run docker load.

By the way, I was looking at `save_docker_cache.sh`. Why is the cache saved only for the master branch?

damonkohler, 2017-05-18:
Thanks for looking into this. I looked the build history and couldn't find the instance where the 0 byte file was saved to the cache. I'm curious to understand what causes that to happen. We don't see this issue frequently, so perhaps it has something to do with your particular Travis configuration?

The workaround is easy enough. Just delete the cache and tell it to rebuild. To identify the issue, I'd suggest modifying your save_docker_cache.sh script to throw an error if the cached file will be 0 size. Then, we can see what lead to that.

ojura, 2017-05-18:
I have been getting this pretty often in the past few months. Cleaning the cache works, but sometimes I have to do it multiple times. I am using a pretty standard Travis configuration, I just forked the repo and enabled Travis. Sometimes I do force push and rebase often, and I have enabled "Auto cancellation" in Travis settings.

damonkohler, 2017-05-18:
I don't have any theories for why you're seeing this so often. I suggest instrumenting save_docker_cache.sh as I described and see what happens.

I forgot to answer your question earlier. We only cache the master branch to avoid having PRs blow away the cache unnecessarily.

ojura, 2017-05-18:
Yeah, that's what I suggested above as well :) I will do that.

I think I managed to reproduce it: push a commit, quickly amend it, and then force push. The build in progress will error out because it can't checkout the previous commit by its ID, since it was amended and force pushed while the build environment for the first build job was booting. It will then create an invalid empty cache which breaks all subsequent builds.

I am not completely sure that the following is what's going on, but I think that loading an invalid cache should definitely not be a show stopper.

PR #298 (2017-05-19): Do not ignore docker history command errors.
Fixes #292
closed 2017-05-19 by damonkohler