cartographer-project/cartographer #302: How to output 4000x4000 size of map like gmapping?

b2220333, 2017-05-20:
Hello~
I have to run a program only can run with 4000x4000 size of map.
Is that possible to change the settings of cartographer?
Thank you~

Entropy512, 2017-05-22:
For 2D, the output is one pixel per unit of TRAJECTORY_BUILDER_2D.submaps.resolution - default is 0.05, if the map is too large you could increase this value.

If the map is too small, you could pad the map with grey pixels up to 4000x4000

b2220333, 2017-05-24:
I found that I should not change the resolution, but I also need the map with 4000x4000.
How to pad the map to 4000x4000?
Which setting that I can use?
Thank you~

Entropy512, 2017-05-25:
Use any image editing software - imagemagick should be able to do it from the command line.  You might have to alter the origin point in the yaml file to compensate - depending on if you care if the coordinate system's origin gets shifted at all.

SirVer, 2017-05-26:
I agree with @Entropy512, this is an image processing task that can be done in post. Cartographer does not help with padding.

closed 2017-05-26 by SirVer