cartographer-project/cartographer #305: ZR300 + Cartograhper ROS 3D-mapping, IMU problem

operationflash, 2017-05-23:
I'm currently using only a zr300 realsense camera and Cartographer ros.
I'm running the following launch files: zr300_nodelet_rgdb.launch (which can be found[ here](https://github.com/intel-ros/realsense/tree/indigo-devel/realsense_camera/launch)) and a launch file containing the following code:
[image]

The configuration lua file contains the following code:
[image]

In terminal I get the following output (and error meessage):
[image]
Unfortunately the error log file doesn't seem to exist.

Does anyone now a solution?

Entropy512, 2017-05-23:
My first guess would be a problem with the timestamps of your IMU data?  As in they should be constantly increasing, but one message had a timestamp earlier than the previous one?

operationflash, 2017-05-23:
@Entropy512 If I look at the IMU data, the time stamps seem to be increasing and in order.

SirVer, 2017-05-26:
@operationflash No, they are not. Cartographer complains about that (see in your last screenshot, there is a CHECK line in the middle in red). When investigating, make sure you look at the `header.stamp` data, not at the timestamp when rosbag wrote this into its file.

closed 2017-05-26 by SirVer

DriftingTimmy, 2017-06-26:
Have you succeeded in running your data in cartographer? Could you please show me your bagfile so I can figure out what I need to create my own bagfile.
@operationflash

operationflash, 2017-06-29:
@DriftingTimmy the problem was solved, @SirVer was right. The IMU very rarely gave wrong timestamps which made it harder to find. This was solved bij using the flat_world_imu_node_main code found in cartographer_turtlebot and modifying it to only filter out bad data.