cartographer-project/cartographer #315: Support localization

wohe, 2017-06-06:
The idea is to have a trajectory builder doing pure localization. Other trajectories contain the map data. To avoid a growing SLAM problem, only a few submaps of the localization trajectory are kept.

This depends on #283. Also see googlecartographer/cartographer_ros#95.

SirVer, 2017-06-22:
I am calling this done. It requires some way of actually enabling it for a trajectory in cartographer_ros, but the functionality is now in Cartographer.

closed 2017-06-22 by SirVer

sradmard, 2017-07-24:
Hi,
I am interested in using cartographer to localize my robot on a preexisting 2D map through ROS. I have already done that with AMCL package in ros, however I would like to get it done with the localizer implemented by cartographer as well. I tried to follow this issue, however I cannot figure out exactly what configurations I need to modify in my .lua file to just activate the localizer. 
I am loading my full map using "map_server" ros package, and have set the following parameters in my .lua file:
TRAJECTORY_BUILDER.pure_localization = true
SPARSE_POSE_GRAPH.optimize_every_n_scans = 20

however, after running the launch file, the submap created by the cartographer and my full map do not match. 
I would really appreciate any advice on this matter.

SirVer, 2017-07-25:
Just continue driving for a bit and it should converge eventually. You should maybe also reduce the `optimize_every_n_scans` further (to 1 or 2).

wohe, 2017-07-25:
@sradmard How did you create the map? If you use Cartographer, you can write out a `.pbstream` file and serve this map passing an argument to the Cartographer ROS node like it is done in the localization demo launch file.

You cannot use `map_server` for serving the map and localize in it with Cartographer. You could write a tool that converts the `.pgm` and `.yaml` to a `.pbstream` file containing a single submap, but it does not exist yet.

sradmard, 2017-07-25:
Thank you for your quick responses. 
@wohe I created the map using Cartographer, but I saved it using "map_server" package by running:
$ rosrun map_server map_saver -f mapname
 It is more clear to me now on what to do. I can run my robot again and create another map, however, I cannot find a way to save the map into a .pbstream file. When I call the following service, it does not provide any file:
$ rosservice call /finish_trajectory 0
Also, calling the following service does not do anything:
$ rosservice call /submap_query 0 1

Am I providing wrong arguments to my service calls, or is there another way to save the map into a .pbstream file.

lanyaye, 2017-07-26:
@sradmard Please call service "write_state" (cartographer_ros_msgs/WriteState) provided by cargotrapher_ROS' node_main to save the map into a .pbstream file.  this API already shows in cartographer_ros's main page.

BrannonKing, 2017-08-01:
When in pure localization mode, does the Cartographer use the incoming IMU and relative position data? Or does it only use the laser scan data when localizing?

sradmard, 2017-08-02:
I have been able to run it with and without IMU in pure localization. However, I realized that when IMU is disabled, you really need to move the robot around very slowly, and with IMU enabled the localization can still catch up with higher speeds. I would still appreciate any feedback from the amazing Google team on the best approach to tune the parameters to speed up the localization.

damonkohler, 2017-08-07:
Localization is just multi-trajectory SLAM. Assuming you map and then localize with the same platform, [the configuration shouldn't change much](https://github.com/googlecartographer/cartographer_fetch/blob/master/cartographer_fetch/configuration_files/freight_localization.lua) and all the same sensor data will be used.

@sradmard if you're having performance issues with localization, please open a new issue.

jpro91, 2017-08-25:
Hi,
This issue is closed, but it looks like that the pure 3D localization is not implemented yet?
By the way, thanks for open sourcing your amazing work.

yinnxinn, 2017-09-23:
@sradmard , may i ask did you get the correct localization just use a 2D radiar ???? if so , could you please give more details about it . i would appreciate if if you could give more info

sradmard, 2017-10-26:
@yinnxinn sorry for the late response. Yes, I got it working with both a depth camera, and just recently with a 2D 360 lidar. I would totally recommend using the lidar option though, as it works greatly. 
If you manage to get the 2D SLAM running, then the Localization should run as well. You just need to:
- load the saved map in your launch file, 
- in your config.lua file set the TRAJECTORY_BUILDER.pure_localization = true,
- in your config.lua file reduce your SPARSE_POSE_GRAPH.optimize_every_n_scans. 

Take a look at demo_backpack_2d_localization.launch under cartographer_ros, and compare it to demo_backpack_2d.launch. That would basically guide you on how to implement your Localization.  
Once again, a big thank to google cartographer team for their amazing work and support.

darthShana, 2020-08-20:
Hi all
Could you provide me with how i can get the localised pose during localization. When using AMCL i was able to listen to amcl_pose. Whats the equivalent in google catographer?

MigVega, 2022-11-27:
Hi all, Hi [@sradmard](https://github.com/sradmard)

Now you can use the [Ogm2Pgbm package](https://github.com/MigVega/Ogm2Pgbm/tree/main/ogm2pgbm) to do the conversion from ` .pgm` to Pose graph based map in ` .pbtream` format !

If haven't compiled cartographer already, you can follow the provided instructions in the [readme](https://github.com/MigVega/Ogm2Pgbm). It will allow you to compile everything in one isolated container.

Otherwise you can also use the package as a normal separated catkin package.

Have fun, and enjoy the power of graph-based localization! :D