cartographer-project/cartographer #320: crash in mapping_2d::CastRays

BrannonKing, 2017-06-07:
I'm seeing the crash (below) repeatedly in my simulation environment. I have a simplified simulation environment where I never see the crash, nor do I see it on my live vehicle. Can somebody explain the limit check? I'm not sure what to look for in my laser data to identify the problematic data. Or did I just configure some limit incorrectly? I'm using a cartographer build of latest source on June 2nd.

```
F0607 10:07:59.000000 164144 probability_grid.h:195] Check failed: limits_.Contains(xy_index) 6081
8034
*** Check failure stack trace: ***
    @     0x7f8b5ac745cd  google::LogMessage::Fail()
    @     0x7f8b5ac76433  google::LogMessage::SendToLog()
    @     0x7f8b5ac7415b  google::LogMessage::Flush()
    @     0x7f8b5ac76e1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x47b4a2  _ZNSt17_Function_handlerIFvRKN5Eigen5ArrayIiLi2ELi1ELi0ELi2ELi1EEEEZNK12cartographer10mapping_2d17RangeDataInserter6InsertERKNS6_6sensor9RangeDataEPNS7_15ProbabilityGridEEUlS4_E_E9_M_invokeERKSt9_Any_dataS4_
    @           0x47c321  cartographer::mapping_2d::CastRays()
    @           0x47b0a3  cartographer::mapping_2d::RangeDataInserter::Insert()
    @           0x535d67  cartographer::mapping_2d::Submaps::InsertRangeData()
    @           0x487437  cartographer::mapping_2d::LocalTrajectoryBuilder::AddHorizontalRangeData()
    @           0x465ade  cartographer::mapping_2d::GlobalTrajectoryBuilder::AddRangefinderData()
    @           0x5147f9  cartographer::mapping::CollatedTrajectoryBuilder::HandleCollatedSensorData()
    @           0x51522e  _ZNSt17_Function_handlerIFvRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt10unique_ptrIN12cartographer6sensor4DataESt14default_deleteISB_EEEZNS9_7mapping25CollatedTrajectoryBuilderC4EPNSA_8CollatorEiRKSt13unordered_setIS5_St4hashIS5_ESt8equal_toIS5_ESaIS5_EES8_INSG_32GlobalTrajectoryBuilderInterfaceESC_IST_EEEUlS7_SE_E_E9_M_invokeERKSt9_Any_dataS7_OSE_
    @           0x49413a  _ZNSt17_Function_handlerIFvSt10unique_ptrIN12cartographer6sensor4DataESt14default_deleteIS3_EEEZNS2_8Collator13AddTrajectoryEiRKSt13unordered_setINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt4hashISF_ESt8equal_toISF_ESaISF_EESt8functionIFvRKSF_S6_EEEUlS6_E_E9_M_invokeERKSt9_Any_dataOS6_
    @           0x48cd69  cartographer::sensor::OrderedMultiQueue::Dispatch()
    @           0x48e30f  cartographer::sensor::OrderedMultiQueue::Add()
    @           0x494434  cartographer::sensor::Collator::AddSensorData()
    @           0x513948  cartographer::mapping::CollatedTrajectoryBuilder::AddSensorData()
    @           0x44617f  cartographer_ros::SensorBridge::HandleOdometryMessage()
```

SirVer, 2017-06-08:
Can you provide data + a commandline for repro?

wohe, 2017-06-08:
In 2D the size of the pixel grid for each submap is configured as `TRAJECTORY_BUILDER_2D.submaps.half_length` which is half the side length in meters. The interesting question is whether you have longer range data or more movement that make this size too small. If it is the case, the solution is to increase this size. If this is not the case, you should see this as an indication that SLAM failed and should be tuned better. Increasing the size in this case could only hide errors which is not a good choice.

BrannonKing, 2017-06-08:
I believe that I did find a problem with the laser data where some corruption would have caused it to be longer than half_length. Thanks for the info.

closed 2017-06-08 by BrannonKing