cartographer-project/cartographer #322: localization in existing map

weisongwen, 2017-06-08:
Hi,  

       Thanks for your excellent work in cartographer. I could create the map using the cartographer with 3D pointCloud2. 

       But, I want to perform the localization in the existing map for a mobile robot, at this case:

1. is there any method that you could recommend ?

2. are your working on the localization function in existing map in terms of the cartographer?

SirVer, 2017-06-09:
We are working on this. Please follow #315.

closed 2017-06-09 by SirVer

weisongwen, 2017-06-10:
Hi, @SirVer 

Thanks  for your reply!

I will track this issue. Looking forward to your excellent work on "localization in existing map".

apart that, could you pls recommend some method else which may able to perform localization in existing 3D map (such as .pcd file )?

Thanks a lot! 
Best regards

SirVer, 2017-06-12:
You could always use [AMCL](http://wiki.ros.org/amcl), but we have little experience with it.

weisongwen, 2017-06-12:
Hi, @SirVer 

AMCL is a package used in 2D robot localization instead of 3D  localization.

Looking forward your progress in cartographer in terms  of localization in existing map.

best regards

SirVer, 2017-06-12:
A sorry, yes that is true. I do not know of a 3D localization package that is open source, sorry.

weisongwen, 2017-06-12:
Hi, @SirVer 

  Thanks a lot anyway, I will try other methods later