cartographer-project/cartographer #325: Support tracking moving landmarks in the SLAM problem

SirVer, 2017-06-09:
Originally by @ahundt in https://github.com/googlecartographer/cartographer/issues/244#issuecomment-307153563. 

Andrew, your original issue contained two points if I paraphrase correctly:

1) the SLAM should be able to support tracking landmarks that are part of the robot, but on movable parts like arms.

2) the SLAM should support tracking landmarks that are known to be moving in the map frame and the movement is known to the user of Cartographer.

Is that correct? I'd like to edit this issue to be your proposal and discuss it here.

ahundt, 2017-06-09:
Yes that captures it nicely! 

Let me consider scope options for a moment:

 - medium: individual landmark entities like an AR tag, which does not modify point clouds
 - large: model based tracking of landmarks so the point cloud can be updated, ex: [DART code](https://github.com/tschmidt23/dart) and [paper](http://mobilerobotics.cs.washington.edu/papers/DepthBasedTracking-icra-2015.pdf)
 - huge: capturing deformation as in [DynamicFusion](https://rse-lab.cs.washington.edu/papers/dynamic-fusion-cvpr-2015.pdf), [VolumeDeform](https://arxiv.org/pdf/1603.08161), and [Fusion4D](www.samehkhamis.com/dou-siggraph2016.pdf)

Hmm, I've been exploring the code a bit and referring to the cartographer paper and the scale of data in papers that track moving entities is much smaller than cartographer typically handles. Perhaps the algorithms could be adapted but they are often quite different. Do the options above fit with the mission of the project? If not, perhaps there is another approach that would fit better?

SirVer, 2017-07-31:
I think none of the above fits Cartographer's mission right now. Modeling moving landmarks is not currently in scope. 

Closing for now.

closed 2017-07-31 by SirVer

ahundt, 2017-08-01:
sensible enough, thanks for the update.