cartographer-project/cartographer #329: Be smarter about what to match in multi trajectory cases

SirVer, 2017-06-12:
This is probably a needed improvement to make #315 really field ready.

Loop closing in single trajectory cases is *quote* smart *unquote* in choosing submaps are tried to can match against - i.e. only those physically close to the local SLAMs opinion of position are tried. This is a desired property to keep computation in check.

In the multi trajectory case, when trajectories are still disconnected, a scan from Trajectory a will be matched against essentially all submaps of trajectory b. This is necessary because there is no knowledge about how the trajectories overlap. However, once some constraints that tie a and b together are found, this is no longer true. With this better knowledge, loop closure should use something akin to the logic for a single trajectory to only search for fruitful matches between a and b.

wohe, 2017-06-26:
One idea would be to keep a state for each pair of trajectories whether the global matcher (match against all submaps) is used or the fast matcher (every submap within a certain distance). When a global match is found, the state changes to the fast matcher for this pair of trajectories. When no matches are found with the fast matcher for some time, the state falls back to the global matcher.

Right now, a trajectory to itself always uses the fast matcher. Different trajectories always use the global matcher, and once they are connected use the fast matcher in addition to that.

Mofef, 2017-09-13:
@cschuet on first glance your recent PRs (e.g. #513 ) indicate that you are working on this issue. Correct?
For us this has also quite high priority, mostly because we would like to eliminate global re-localization once a correct localization was confirmed. 
So is there still anything we could help with?

Mofef, 2017-09-15:
@wohe you might want to close this because it is solved after #517

closed 2017-09-15 by wohe