cartographer-project/cartographer #362: How to configure the .lua file and the programe to run my own 3D data with imu?

DriftingTimmy, 2017-06-26:
I have a VLP16 lidar and an imu can provide the needed data, but I can't figure out how to change the corresponding .lua file to make the cartographer run.
And when I make my own bagfile, I can't make the frequency of pointcloud2 that speed as nearly 5 times than imu data.

SirVer, 2017-06-26:
See the [FAQ](https://google-cartographer-ros.readthedocs.io/en/latest/faq.html#the-laser-data-in-the-3d-bags-is-much-higher-than-the-maximum-reported-20-hz-rotation-speed-that-the-vlp-16-can-do-why) for the frequency explanation. There is a chance that you need to change `TRAJECTORY_BUILDER_3D.scans_per_accumulation = 160` from 160 to 1 in your setup.

For the other questions: please provide configuration, launch files and a bag file.

closed 2017-06-26 by SirVer

DriftingTimmy, 2017-06-26:
@SirVer  ok, I will give those files in a few days. Thanks very much!