cartographer-project/cartographer #377: How can I modify the configuration file to run 3D slam with only one horizontal_laser_3d?

DriftingTimmy, 2017-06-30:
I want to use the 3D slam method with the data from the backpack demo's  bagfile. So I use "rosbag record -O test1.bag /imu /horizontal_laser_3d" to record data from "b3-2016-04-05-14-14-00.bag" and I get the bagfile for about 15 seconds data. The data in the bagfile can be shown like below:
[image]

Then I modify the configuration file with my own knowledge (something must be wrong cause I m new to this....)
[configuration file.tar.gz](https://github.com/googlecartographer/cartographer/files/1114022/configuration.file.tar.gz)

Then when I run "roslaunch cartographer_ros demo_test.launch bag_filename:=${HOME}/Downloads/test1.bag" I get wrong like this:
[image]

And I run "rqt_graph" to check the messages and topics, everything seems to be ok.
[image]

So what should I do to run the programme well with only one horizontal lidar data and imu?

SirVer, 2017-07-03:
Please do not make screenshots of text terminals. Instead, provide the full textual output of your roslaunch command as a gist.

You did not provide your bag, so I can only guess without the means of investigating a repro case. I guess that your `frame_id` in your laser topic is not `horizontal_vlp16_link`. You can test this by running `rostopic echo -n1 /horizontal_laser_3d | less` and look for the `frame_id`.

closed 2017-07-05 by SirVer

DriftingTimmy, 2017-07-06:
Sorry for applying u so late. I was trying to find the way to solve it but the error occurred everytimes. When I run my own launch file, the output is:

... logging to /home/timmy/.ros/log/aee3e8b0-61e8-11e7-8e0b-1c1b0d00bd07/roslaunch-timmy-B85M-D3H-6299.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://timmy-B85M-D3H:45412/

SUMMARY
========

PARAMETERS
 * /robot_description: <...>
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    playbag (rosbag/play)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[robot_state_publisher-1]: started with pid [6317]
process[cartographer_node-2]: started with pid [6318]
process[rviz-3]: started with pid [6339]
process[playbag-4]: started with pid [6340]
[ INFO] [1499304507.402099724]: I0706 09:28:27.000000  6318 configuration_file_resolver.cc:40] Found '/home/timmy/catkin_ws3/install_isolated/share/cartographer_ros/configuration_files/my_data.lua' for 'my_data.lua'.
[ INFO] [1499304507.402439016]: I0706 09:28:27.000000  6318 configuration_file_resolver.cc:40] Found '/home/timmy/catkin_ws3/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1499304507.402514857]: I0706 09:28:27.000000  6318 configuration_file_resolver.cc:40] Found '/home/timmy/catkin_ws3/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1499304507.402648274]: I0706 09:28:27.000000  6318 configuration_file_resolver.cc:40] Found '/home/timmy/catkin_ws3/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1499304507.402721405]: I0706 09:28:27.000000  6318 configuration_file_resolver.cc:40] Found '/home/timmy/catkin_ws3/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1499304507.402916746]: I0706 09:28:27.000000  6318 configuration_file_resolver.cc:40] Found '/home/timmy/catkin_ws3/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1499304507.402989344]: I0706 09:28:27.000000  6318 configuration_file_resolver.cc:40] Found '/home/timmy/catkin_ws3/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1499304507.403112147]: I0706 09:28:27.000000  6318 configuration_file_resolver.cc:40] Found '/home/timmy/catkin_ws3/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1499304507.403181608]: I0706 09:28:27.000000  6318 configuration_file_resolver.cc:40] Found '/home/timmy/catkin_ws3/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1499304507.403354734]: I0706 09:28:27.000000  6318 configuration_file_resolver.cc:40] Found '/home/timmy/catkin_ws3/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1499304507.403428959]: I0706 09:28:27.000000  6318 configuration_file_resolver.cc:40] Found '/home/timmy/catkin_ws3/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1499304507.424077786]: I0706 09:28:27.000000  6318 submaps.cc:410] Added submap 1
[ INFO] [1499304507.424194420]: I0706 09:28:27.000000  6318 map_builder_bridge.cc:40] Added trajectory with ID '0'.
[ WARN] [1499304507.706565533, 1498699284.338775344]: W0706 09:28:27.000000  6318 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1459865641.519203200 but the earliest data is at time 1498699284.828712132, when looking up transform from frame [horizontal_vlp16_link] to frame [base_link]
[ WARN] [1499304507.706697255, 1498699284.338775344]: W0706 09:28:27.000000  6318 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1459865641.519924000 but the earliest data is at time 1498699284.828712132, when looking up transform from frame [horizontal_vlp16_link] to frame [base_link]
[ WARN] [1499304507.706766895, 1498699284.338775344]: W0706 09:28:27.000000  6318 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1459865641.520529900 but the earliest data is at time 1498699284.828712132, when looking up transform from frame [horizontal_vlp16_link] to frame [base_link]
[ WARN] [1499304507.706838727, 1498699284.338775344]: W0706 09:28:27.000000  6318 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1459865641.521160900 but the earliest data is at time 1498699284.828712132, when looking up transform from frame [horizontal_vlp16_link] to frame [base_link]
[ WARN] [1499304507.706912953, 1498699284.338775344]: W0706 09:28:27.000000  6318 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1459865641.521890100 but the earliest data is at time 1498699284.828712132, when looking up transform from frame [horizontal_vlp16_link] to frame [base_link]
[... cut ...] se_link]

And I checked the frame_id you have mentioned that I think the output is ok.

header: 
  seq: 0
  stamp: 
    secs: 1459865647
    nsecs: 277518600
  frame_id: horizontal_vlp16_link
height: 1
width: 384
fields: 
  - 
    name: x
    offset: 0
    datatype: 7
    count: 1
  - 
    name: y
    offset: 4
    datatype: 7
    count: 1
  - 
    name: z
    offset: 8
    datatype: 7
    count: 1
is_bigendian: False
point_step: 16
row_step: 6144

And I will upload my bagfile soon,thx very much

DriftingTimmy, 2017-07-06:
https://pan.baidu.com/s/1pLTtZqr
I just upload my bagfile, but I think the data in the file is just the same as the data in the demo bag.
Thx.

SirVer, 2017-07-06:
> https://pan.baidu.com/s/1pLTtZqr

This leads me to a page does not exist page. 

> [ WARN] [1499304507.707116705, 1498699284.338775344]: W0706 09:28:27.000000 6318 tf_bridge.cc:51] Lookup would require extrapolation into the past. Requested time 1459865641.524512200 but the [...]

This seems like your URDF is not correct, not loaded or your bag has different frame ids. I cannot be sure.

DriftingTimmy, 2017-07-10:
I've just sent an email to you which contains my bagfile. I checked the urdf and I don't know what's wrong, there just can not show the map.
When I tried to see the topic output from the cartographer_node, I get nothing:

timmy@timmy-B85M-D3H:~/catkin_ws3$ rostopic echo /scan_matched_points2 
WARNING: no messages received and simulated time is active.
Is /clock being published?

It seems that no data was accepted by the cartographer node, but the /horizontal_laser_3d topic is published by the bag and I can see it in rviz.

@SirVer  Thanks very much!

damonkohler, 2017-07-10:
Are you sure [use_sim_time](http://wiki.ros.org/Clock) is enabled?

DriftingTimmy, 2017-07-10:
I set the "/use_sim_time" value to be true. And when I set this to false, I need to play my bagfile in another terminal.
Do you mean that the cartographer_node can't receive the data from the bagfile because of the value of the "/use_sim_time" is false?

damonkohler, 2017-07-10:
The error messages made me think you may not have /use_sim_time enabled. If the sensor time stamps are far removed from the time that ROS is using, you can get TF errors that look like the ones you're seeing.

DriftingTimmy, 2017-07-10:
I can understand your meaning a little. But it is just a warning, it seems that everything else goes wrong except the rosbag play. So what should I do to make this run correctly?

SirVer, 2017-07-10:
I investigated the bag and configuration files you posted and I found that the timing of your sensors is terrible. You cannot lookup transforms because the jitter in your data timing is so big - see attached screenshot from rqt_bag. You need to fix your data. The configuration files seemed correct to me.

If you run `cartogarpher_offline_node` instead of the online node you will get a map, since we are not bound by ROS concept of timouts there. But you cannot expect good results with this kind of bad timing.

[image]

DriftingTimmy, 2017-07-10:
Thank you so much!
But why this happened? I just made the bagfile using "rosbag record -O" command to record the topic "/horizontal_laser_3d" and "/imu" in the "b3-2016-04-05-14-14-00.bag" you posted as the demo. How can I divide the data from the demo bagfile to run the cartographer with only one horizontal lidar?
Thanks again~

DriftingTimmy, 2017-07-10:
I just use rqt_bag to compare the data in the demo bagfile and the bagfile I recorded and I found that you are right about this. The data I recorded from the demo is just like a %&&... 
[image]
The demo data is like above, the recorded is just like following:
[image]

SirVer, 2017-07-10:
You can try to filter the bags using a small python script: http://wiki.ros.org/rosbag/Cookbook

But you will not be able to run 3D SLAM with only the horizontal laser in the deutsche museum. The opening angle of the laser is not big enough to consistently see floor or ceiling, so the SLAM problem is not constraint in z-direction.

DriftingTimmy, 2017-07-10:
ok~ I got it~ Thanks for explaining.
So if I want to use my own lidar data and imu data, this kind of configuration will work, right? I think some parameters should be changed but most of the configuration is ok.
Cause I haven't got my own lidar data right now, I tried in this way to make sure the configuration is alright, when I get my own data, I will post my result here.
Thanks a lot~

SirVer, 2017-07-10:
> So if I want to use my own lidar data and imu data, this kind of configuration will work, right? 

It mostly depends on your surroundings and your lidar. You need to see enough to constrain your problem in all dimensions.

Good luck!