cartographer-project/cartographer #390: Trimmer crashes

jihoonl, 2017-07-05:
As @Mofef mentioned in #389, the latest code crashes while trimming submaps. Here is the error log.

```
[/cartographer_offline_node  INFO 1499262071.746585000, 1495552307.184251417]: I0705 15:41:11.000000 14339 constraint_builder.cc:245] Node (0, 157) with 103 points on submap (0, 4) differs by translation 0.00 rotation 0.001 with score 62.4%.
[/cartographer_offline_node  INFO 1499262071.746623421, 1495552307.184251417]: I0705 15:41:11.000000 14339 constraint_builder.cc:266] 22 computations resulted in 22 additional constraints.
[/cartographer_offline_node  INFO 1499262071.746768683, 1495552307.184251417]: I0705 15:41:11.000000 14339 constraint_builder.cc:268] Score histogram:
Count: 24  Min: 0.465051  Max: 0.696167  Mean: 0.563143
[0.465051, 0.488163)	                  ##	Count: 2 (8.333333%)	Total: 2 (8.333333%)
[0.488163, 0.511274)	                  ##	Count: 2 (8.333333%)	Total: 4 (16.666666%)
[0.511274, 0.534386)	                 ###	Count: 4 (16.666666%)	Total: 8 (33.333332%)
[0.534386, 0.557497)	                 ###	Count: 3 (12.500000%)	Total: 11 (45.833332%)
[0.557497, 0.580609)	                ####	Count: 5 (20.833334%)	Total: 16 (66.666664%)
[0.580609, 0.603721)	                 ###	Count: 3 (12.500000%)	Total: 19 (79.166664%)
[0.603721, 0.626832)	                   #	Count: 1 (4.166667%)	Total: 20 (83.333336%)
[0.626832, 0.649944)	                  ##	Count: 2 (8.333333%)	Total: 22 (91.666664%)
[0.649944, 0.673056)	                   #	Count: 1 (4.166667%)	Total: 23 (95.833336%)
[0.673056, 0.696167]	                   #	Count: 1 (4.166667%)	Total: 24 (100.000000%)
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
[cartographer_offline_node-3] process has died [pid 14314, exit code -6, cmd /home/jihoonl/work/ros/localization/install/lib/cartographer_ros/cartographer_offline_node -configuration_directory /home/jihoonl/work/ros/simws/src/toru_rosbag_play/config/cartographer -configuration_basename toru.lua -bag_filenames /home/jihoonl/work/rosbag/test.bag -urdf_filename /home/jihoonl/work/ros/simws/src/toru_rosbag_play/config/urdf/toru_for_carto.xacro scan:=/scan_aggregated __name:=cartographer_offline_node __log:=/home/jihoonl/.ros/log/97b6dd52-6187-11e7-8208-28d2442d0551/cartographer_offline_node-3.log].
log file: /home/jihoonl/.ros/log/97b6dd52-6187-11e7-8208-28d2442d0551/cartographer_offline_node-3*.log
```

And I have quickly confirmed that it is not crashing in the version 7f4f78e4513d583b4a1a90f76014db26871cf49e. So, I assume that one of refactoring work broke the trimming process.

jihoonl, 2017-07-05:
I have confirmed that d8eb94b4a92d7aeb84c43a73f94e7905eb17148f causes the crash.

SirVer, 2017-07-05:
Cannot reproduce. Works for me at head. Have you nuked your workspace and rebuild from scratch? Never trust catkin.

SirVer, 2017-07-05:
Please try running the demo as documented in  https://github.com/googlecartographer/cartographer_ros/pull/410 once this PR is merged. I do not see a crash with this. As mentioned above, I believe you have a build issue. Closing, if you disagree we can reopen.

closed 2017-07-05 by SirVer

jihoonl, 2017-07-05:
I tried again after rebuilding from scratch. And it was still crashing. Perhaps, trimming doesn't support the single trajectory case? Or `trajectory_id = 0` being trimmed?. I will report back after testing the demo in #410.

jihoonl, 2017-07-06:
The problem still remains with our rosbag. But, I couldn't reproduce this error with backpack_demo.

SirVer, 2017-07-06:
@jihoonl Are you investigating yourself? Otherwise you need to make config and bag available to us somehow.

jihoonl, 2017-07-06:
I am investigating myself for now. I will post if I need a help.

jihoonl, 2017-07-06:
It was because our robot does not have imu data.

PR #392 (2017-07-06): Check imu_data array size before access
To resolve #390.