cartographer-project/cartographer #402: Questions about using RealTimeCorrelativeScanMatcher for odometry

myu20, 2017-07-10:
I am trying to use `RealTimeCorrelativeScanMatcher` in `mapping_3d/scan_matching` with point clouds to get odometry, as was suggested in [this post](https://github.com/googlecartographer/cartographer_ros/issues/305).

1. Is the updated `pose_estimate` provided by the scan matcher the relative pose between the current scan and the reference scan?

2. How do I render a scan into a `HybridGrid`? Do I loop through the points and call `SetProbability` and `GetCellIndex` as in the [test code](https://github.com/googlecartographer/cartographer/blob/9033fad1aba3cd975dc882d80f8d604229450e24/cartographer/mapping_3d/scan_matching/real_time_correlative_scan_matcher_test.cc#L42)?

3. What do I use as the `initial_pose_estimate`? The pose estimate from the previous scan?

Thank you!

wohe, 2017-07-11:
I think googlecartographer/cartographer_ros#305 was about 2D. I understand you want to use the 3D code which is slightly different and there is less experience with it.

1. `initial_pose_estimate` and `pose_estimate` are both in the frame of the hybrid grid. You can of course choose that frame to be the frame of the reference scan by inserting the reference scan appropriately.

2. No need to do this, you can use the `RangeDataInserter`.

3. No. It is your best guess of `pose_estimate`. If you insert the untransformed reference scan, a guess could for example be identity if you think not much has changed between the two scans.

I hope that helps. Let me know if you have further questions.

myu20, 2017-07-11:
Thank you for your reply! I am confused about `pose_estimate`. If I render a point cloud `a` into a hybrid grid, then call `Match` with the hybrid grid and a point cloud `b` that is later in time, is `pose_estimate` the relative pose of `b` with respect to `a`?

wohe, 2017-07-11:
Yes, unless you transform `a` before handing it to the `RangeDataInserter` this should be the result.

myu20, 2017-07-11:
Ok, thank you!

closed 2017-07-24 by SirVer