cartographer-project/cartographer #404: Saving and loading submaps

GMahmoud, 2017-07-13:
Hi everybody,

I am testing cartographer for a 2D mapping. And I wanna know if it is it possible to save submaps at the end of a running bagfile and load them after that as initial submaps when I re-play the same to test the pure localization.

Thank you for your answer 
Cheers,
MG

brandon-northcutt, 2017-07-13:
@GMahmoud It is possible to save the cartographer state as needed for pure localization mode. The instructions are in the cartographer_ros [Read The Docs site](https://google-cartographer-ros.readthedocs.io/en/latest/). You should be able to test pure localization mode against your own bag files using similar commands. Full serialization as needed to resume and add to the state is still in the works #253.
```BASH
# Pure localization demo: We use 2 different 2D bags from the Deutsche
# Museum. The first one is used to generate the map, the second to run
# pure localization.
wget -P ~/Downloads https://storage.googleapis.com/cartographer-public-data/bags/backpack_2d/b2-2016-04-05-14-44-52.bag
wget -P ~/Downloads https://storage.googleapis.com/cartographer-public-data/bags/backpack_2d/b2-2016-04-27-12-31-41.bag
# Generate the map: Run the next command, wait until cartographer_offline_node finishes.
roslaunch cartographer_ros offline_backpack_2d.launch bag_filenames:=${HOME}/Downloads/b2-2016-04-05-14-44-52.bag
# Run pure localization:
roslaunch cartographer_ros demo_backpack_2d_localization.launch \
   bag_filename:=${HOME}/Downloads/b2-2016-04-27-12-31-41.bag \
   map_filename:=${HOME}/Downloads/b2-2016-04-05-14-44-52.bag.pbstream
```

GMahmoud, 2017-07-14:
Perfect @brandon-northcutt and thank you guys for your amazing work !

closed 2017-07-14 by GMahmoud