cartographer-project/cartographer #409: Trimming a submap in the middle

jihoonl, 2017-07-18:
This is a necessary work to replace an outdated submap with a newly generated submap that covers the same area. With this feature, I think cartographer can actually do the lifelong mapping over time. I would like to open a discussion to start the development.

I think we can break this down into two separate problems, trimming trajectory nodes and trimming a submap.

Submap trimming seems relatively easy. All we have to do is changing vector<deque> to vector<map> just as in cartographer_rviz (and of course some following works).

Trimming trajectory nodes seems to need a design decision(or discussion) to decide how cartographer keeps the separated trajectories. Should cartographer keep the separated trajectories in the nested map structure? should it keep the entire trajectory without trimming until no constraint references this trajectory? Or any other options?

How do you think?

SirVer, 2017-07-18:
We thought that this should be modeled by 

1) Split trajectory.
2) Trim one of the from the beginning.

2) is done, 1) should be relatively simple (famous last words).

If you trim nodes in the middle you have an awkward trajectory with missing data and non-even timestamp increases.

jihoonl, 2017-07-21:
Alright. I will try to split trajectory.

wohe, 2017-08-09:
I was trying to review #425, but I think we need some discussion first how we approach trimming in the middle. How I see it, there are a couple of things that matter:

- we label data presented to the user with pairs of trajectory ID and node index (e.g. in RViz) where it is useful to look at individual trajectories (similarly we use ID/index pairs for submaps)
- we also use these ID pairs to label input data, i.e. to tell the SLAM system which trajectory builder some data is for
- we need to be able to trim arbitrary nodes and submaps for life-long mapping, and this should be efficient
- the internal representation of the pose graph is an implementation detail, therefore it does not have to use ID/index pairs

I see two very different general approaches to implement trimming in the middle:
1. Change the so called "trajectory IDs" to only identify where data came from and allow "trajectories" to contain holes. "Trajectories" with holes are no longer a single trajectory. We must detect and handle these gaps in node indices, e.g. in RViz and SLAM. By construction this maintains labels, e.g. in RViz it is easy for users to understand what is happening. In this case, the data structures for nodes/submaps could be changed from `std::vector<std::deque<..>> to `std::vector<std::map<int,...>>`.
2. The pose graph becomes just a graph structure which does not contain labels, for input we keep track of then end of each trajectory so trajectory builders can attach new nodes/submaps. Whenever we present data to the user, we label data. For stability, e.g. in RViz, we might also keep some additional data for that purpose. Most importantly, when a trajectory is split no relabeling happens, i.e. removing every other node does not become O(n^2).

1 seems preferable to me. What do you think? Interesting alternative approaches I forgot?

jihoonl, 2017-08-10:
I agree with your point that “cartographer should be able to trim arbitrary nodes and submaps for life-long mapping”.

> Change the so called "trajectory IDs" to only identify where data came from and allow "trajectories" to contain holes. "Trajectories" with holes are no longer a single trajectory. We must detect and [...]

I actually have an experimental branch that I attempted to redesign the sparse pose graph to allow arbitrary submap and nodes trimming. See [spare_pose_graph.cc](https://github.com/magazino/cartographer/blob/better_trimmer/cartographer/mapping_2d/sparse_pose_graph.cc#L605) and [pose_grapher_trimmer.cc](https://github.com/magazino/cartographer/blob/better_trimmer/cartographer/mapping/pose_graph_trimmer.cc#L30) in [better_trimmer branch](https://github.com/magazino/cartographer/tree/better_trimmer). I changed the data structure of submap to `std::vector<map>` and separated MarkSubmapAsTrimmed into two functions to handle submap trimming and node trimming independently to allow a trimmer to have more control to determine which submaps and which nodes to trim.

To manage the gaps created by trimmed trajectory nodes, one of my idea was introducing `std::vector<std::map<int, std::vector<TrajectoryNode>>>` to keep split trajectory separately within the same trajectory id. Or, we can simply introduce a logic in [optimization_problem.cc](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping_2d/sparse_pose_graph/optimization_problem.cc#L195) to detect and handle the gaps created by trimmed trajectory nodes. 

I decided not to continue in this direction after @Sirver’s comment because it seemed to introduce a major data structure change which you may not like. If you consider this direction is worth to discuss, we can continue further.

> The pose graph becomes just a graph structure which does not contain labels, for input we keep track of then end of each trajectory so trajectory builders can attach new nodes/submaps. Whenever we [...]

One of limiting factor of the current data representation is that cartographer uses ID pair(trajectory id, index) as a unique identifier for both submaps and trajectory nodes. If we introduce another way of a unique identifier for each submaps and trajectory nodes, relabelling will not happen.

Option 1 seems relatively easy to do. So, I would also prefer option1 unless option 2 has significant benefit. Shall I revive #422 ? :)

About #425, I think it is still a valid approach regardless of this discussion. Sparse pose graph uses trajectory id as an important unique identifier to maintain its internal data structure, and map builder and trajectory builders use this trajectory id just to feed sparse pose graph. So, I think the trajectory id creation should be done within sparse pose graph since trajectory id is designed for sparse pose graph. Also, there might be a use case that having multiple map builders that share one sparse pose graph?

wohe, 2017-08-11:
Your experimental branch seems to contain multiple changes. Let's focus on getting the data structure right first, and let's start with submaps first. As I see it this would also mean removing the `num_trimmed_submaps` which are no longer needed as this data is included in the `map<int, ...>`.

Regarding the `std::vector<std::map<int, std::vector<TrajectoryNode>>>`, I think you cannot make this efficient. It can happen that you have to split the innermost `std::vector<>` many times leading to quadratic behavior. Logic to detect and handle the gaps is a better idea in my opinion.

#422: Yes, but it should probably remove the `num_trimmed_submaps` at the same time.

Re #425: IDs are used to get input data into the sparse pose graph. I do not think the sparse pose graph should have control over these IDs.

johnzjq, 2017-08-21:
I have tried MarkSubmapAsTrimmed in the latest code #462. It worked just fine when I removed several Finished submaps. I am planning to insert the Trimmed stuff (submap, constraints, Node) later to see the outcome.

johnzjq, 2017-08-21:
well, trimming still needs to operate in orders, rather than from middle....I will try your solution @jihoonl

johnzjq, 2017-08-21:
I planned to dynamically paging submaps and trajectory data between memory and disk. However, this would require the triming of arbitrary submaps and further restoring the once trimmed submaps. Now triming arbitrary submaps seems already a headache problem... I am not sure whether splitting trajectory would work in my case.

johnzjq, 2017-08-22:
Is it possible by adding a flag in the submap struct, and then trimming the probability grids of submaps with certain flag instead? The trajectory will not be broken.

jihoonl, 2017-09-13:
I think this issue is resolved via #472, #422

closed 2017-09-13 by jihoonl