cartographer-project/cartographer #410: directly include gmock?

ahundt, 2017-07-18:
I'm on OS X and I was trying to build cartographer, however they seem to recommend incorporating it directly in the project build:
https://github.com/sourcegraph/homebrew/blob/master/Library/Homebrew/blacklist.rb#L42

There are a few options to incorporate it directly:

1. This could probably be done in cmake with ExternalProject_Add when gmock isn't found. 
2. https://github.com/ruslo/hunter supports it
3. the code could be committed directly in a third_party folder

Could any of these options be considered?

SirVer, 2017-07-19:
1. seems to be the most generic option. I would not want to depend on hunter, since the industry does not seem to converge around a single package manager I think it is best to avoid them all together for now. 

The 4. approach I do not understand. Do they just maintain a brew formula and that's it? It seems like there are no gRPC hacks in the repo.

> Also it seems googlemock has been absorbed into googletest https://github.com/google/googletest

Yes. Unfortunately Cartographer still has to deal with systems where they are two separate libraries, so we cannot assume this to be always true while building.

gaschler, 2017-12-07:
Closing for inactivity, feel free to re-open if needed.

closed 2017-12-07 by gaschler

ahundt, 2017-12-07:
yes, they just have a brew formula that does the configuration + install. @gaschler I'm not sure 1 has been completed so I believe this should remain open...