cartographer-project/cartographer #412: Control reaches end of non-void function in test

ojura, 2017-07-19:
When compiling with `CMAKE_BUILD_TYPE=Debug` (yikes, yes, I know), I get the following error:

```
catkin_ws/src/cartographer/cartographer/io/hybrid_grid_points_processor.cc:64:1: error: control reaches end of non-void function [-Werror=return-type]
```

I have no idea why building regularly (with `CMAKE_BUILD_TYPE=Release`) doesn't trigger this.

ojura, 2017-07-19:
ping @jordansmithgoogle :-)

damonkohler, 2017-07-20:
Yes, we ran into this issue the other day with #405. It has to do with the compiler flags. I'm in favor of simply returning a value after LOG(FATAL). @SirVer or @wohe, thoughts?

SirVer, 2017-07-20:
For widelands, we use a [`NEVER_HERE`](https://github.com/widelands/widelands/blob/a4a4644650accf46030b8c64963f59a2388637f5/src/base/wexception.h#L63) macro to indicate this. But it seems just silly, since `LOG(FATAL)` is exactly this. Maybe `error=return-type` is disabled in release mode?

ojura, 2017-07-21:
> Maybe error=return-type is disabled in release mode?

I have looked at the full g++ invocations; that's not it. `-Werror=return-type` is present in both Debug and Release builds.

The exact thing that impacts this is `-O2`. If it's there, it compiles. If it isn't, the warning/error is triggered. Perhaps this is a gcc bug?

ojura, 2017-07-21:
After a bit more testing, as I suspected above, I have concluded that this is really weird behaviour of gcc , and have submitted a bug at https://gcc.gnu.org/bugzilla/show_bug.cgi?id=81508.

ojura, 2017-07-22:
Just a quick update, Jonathan Wakely over at the gcc bugzilla has confirmed that this is indeed a bug in GCC 6.

SirVer, 2017-07-24:
Is this worth fixing then? Is gcc6 widely used *today* or will a patch roll out quickly enough? I can imagine many work arounds on our side, but I'd rather not pollute our code base for a single compiler bug if the benefits are not substantial.

ojura, 2017-07-24:
GCC6 is the default C++ compiler on Ubuntu 16.04 up to (at least) 17.10. I would say that it is worth fixing, because I occasionally do use the debug build to take a look at the Cartographer internals and it can quite annoying that it fails to build because of this.

SirVer, 2017-07-24:
@ojura Okay, but this is not the only place where we have this pattern. Why does it not fail elsewhere. Other places are for example:

- ConfigurationFileResolver::GetFullPathOrDie
- DummyFileResolver
- mapping::ToProto

and more. Do you know why this does not bother the compiler?

ojura, 2017-07-24:
Jonathan Wakely has explained that in order for the bug to be triggered, there has to be a local variable (e.g. a string) with a non-trivial destructor, and it has to be destructed after LOG(FATAL). `ConfigurationFileResolver::GetFullPathOrDie` has all variables within the for loop, so the destruction is not after LOG(FATAL). DummyFileResolver doesn't have any local variables.

SirVer, 2017-07-24:
Sorry, I missed the link somehow. So this is indeed the only place where this bug triggers in the code base for now.

PR #418 (2017-07-25): Fix debug build
Add a fake return value after LOG(FATAL) in
HybridGridPointsProcessor::Flush(). This keeps gcc versions lower
than 7 happy when building without optimisation (i.e. debug build).
Fixes #412.
commit 347dc8ce (2017-07-25): Fix debug build (#418)

Add a fake return value after LOG(FATAL) in
HybridGridPointsProcessor::Flush(). This keeps gcc versions lower
than 7 happy when building without optimisation (i.e. debug build).
Fixes #412.
closed 2017-07-25 by wohe