cartographer-project/cartographer #414: ProbabilityPointsProcessor is mirroring the output

SirVer, 2017-07-20:
Compare XRayPointsProcessors output to ProbabilityGridPointsProcessor. The second one is flipping its output horizontally which make the map incorrect.

@damienrg Are you interested in looking into this?

damienrg, 2017-07-20:
I can look at this. Could you post images of XRayPointsProcessor and ProbabilityGridPointsProcessor showing the problem?

SirVer, 2017-07-21:
Yup, here are the images. I created them using the changes in [this commit](https://github.com/SirVer/cartographer_ros/commit/67a270e2aa2ec019b5274a5ec928adc4ed4ac195). First, I created the `.pbstream` using:

~~~
roslaunch cartographer_ros offline_backpack_2d.launch \
    bag_filenames:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag
~~~

then generating the assets using

~~~
roslaunch cartographer_ros assets_writer_backpack_2d.launch \
   bag_filenames:=$HOME/Downloads/cartographer_paper_deutsches_museum.bag  \
   pose_graph_filename:=$HOME/Downloads/cartographer_paper_deutsches_museum.bag.pbstream
~~~

The assets will be written into `~/.ros`. Notice how the probability grid is differently rotated and also flipped over. The X-Ray represents the world as watched from the top, which is also what RViz and ROS are using, so the probability grid should follow this convention.

[image]
[image]

commit 21cc2706 (2017-07-24): Fix ProbabilityPointsProcessor mirroring output (#419)

Fixes #414.
closed 2017-07-24 by SirVer