cartographer-project/cartographer #426: proto_stream.cc cause LoadMap() failed in 32bit platform such as raspberry

lanyaye, 2017-07-25:
Currently I meet one issue on 32bit ARM platform to loadMAP which generated previously.
the cartographer_ros log shows 

> FATAL "Check failed: " "reader->ReadProto(&pose_graph)

I know that cartographer is suggested to run on 64bit platform but sometimes we need try it on 32bit platform such as raspberry.
the issue is in proto_stream.cc

`// First eight bytes to identify our proto stream format.`
`const size_t kMagic = 0x7b1d1f7b5bf501db;`

`void WriteSizeAsLittleEndian(size_t size, std::ostream* out) {`
`  for (int i = 0; i != 8; ++i) {`
`    out->put(size & 0xff);`
`    size >>= 8;`
`  }`
`}`

`bool ReadSizeAsLittleEndian(std::istream* in, size_t* size) {`
`  *size = 0;`
`  for (int i = 0; i != 8; ++i) {`
`    *size >>= 8;`
`    *size += static_cast<size_t>(in->get()) << 56;`
`  }`
`  return !in->fail();`
`}`

here it use the type of "size_t" to identify the 8 bytes data.
but in most 32bit platform,the compiler will defined it("size_t") as 4 bytes length instead of 8 bytes.
this will cause ReadSizeAsLittleEndian always return 0 and kMagic is truncated to 4byte.

I don't know whether we can change "size_t" in file proto_stream.cc to "long long unsigned int" or other 64bit type?

thank you!

wohe, 2017-07-25:
Yes, that is possible. The appropriate 64-bit unsigned integral type would be `uint64` from `common/port.h`. Do you want to give it a try and prepare a PR?

lanyaye, 2017-07-25:
I have tested uint64 from common/port.h in 32bit platform, it works and I have prepared a PR, please review. thank you!

commit 56dbad15 (2017-07-25): Fix proto_stream to support 32-bit platforms. (#427)

In proto_stream.cc, size_t is only correct on 64-bit platforms. Changing it to uint64 adds support for 32-bit platforms. This fixes #426.
closed 2017-07-25 by wohe