cartographer-project/cartographer #431: read proto_stream in python?

ahundt, 2017-07-26:
How feasible is it to read the proto stream in python? I'd like to load and visualize some data.

SirVer, 2017-07-27:
You'd need to wrap or reimplement [ProtoStream](https://github.com/googlecartographer/cartographer/blob/master/cartographer/io/proto_stream.h) and add functionality to get the binary blob so you could decode it in python yourself. 

The better solution is to add the visualization need you have to our rViz plugin. Less work is to do a data export into something you can easily parse with python.

ahundt, 2017-07-27:
What about PR with two additional fixed header protos: One would be a file header message with a (new) magic number, and the other a message header with a size? It could be implemented to use the old magic number to be backwards compatible in C++ and easier to implement in a new language with the new version.

SirVer, 2017-07-28:
I did not understand what you were proposing in your last comment. Could you clarifyS

ahundt, 2017-07-29:
Right now it looks like the data format is something like the following written as binary data:
```
[magic_number][size][message.proto][size][message.proto][size][message.proto]...
```

I am proposing:

```
message ProtoStreamMessageSize {
fixed64 size = 1;
}

message ProtoStreamHeader {
// magic number identifying the proto stream
fixed64 id = 1;
fixed64 version = 2;
}
```

Where the very first one contains a magic number and all others contain size:
```
[ProtoStreamHeader.proto][ProtoStreamMessageSize.proto][Message.proto][ProtoStreamMessageSize][Message.proto][ProtoStreamMessageSize][Message.proto]...
```

In other words, WriteSizeAsLittleEndian would be updated to write a proto containing a single fixed size entry at the beginning and before each message. 

For backwards compatibility: 

1. the existing code would be kept in place
2. if the original magic number doesn't match then
3. try reading the bytes as a protobuf
4. if the magic number does match, read as this new updated format

This updated setup should be quite easy to implement for any data saved with version 2 or higher (where the current code is version 1).

SirVer, 2017-07-31:
sorry, I must be dense today, but I do not get it. What is the advantage of this encoding scheme? It seems strictly more complex, but with the same features to me.

wohe, 2017-08-01:
I think it also does not work. How would you know the size of the protos? For example a valid `ProtoStreamMessageSize.proto` would be the empty string to encode a zero size. How would you know? Do you suggest guessing the message size? That seems wrong to me.

ahundt, 2017-08-01:
ah... I didn't realize [fixed64 is a misnomer](https://github.com/google/protobuf/issues/1200)... If only they called it `nearlyButNotActuallyFixed64`. :-)

So the workaround would be to use `sfixed64` where negative values map to 0, and 0 is never written. Thoughts?

wohe, 2017-08-01:
I don't think I understand what you are trying to achieve. What benefit would this give us? Holger already commented on the disadvantage of such an implementation: unnecessary complexity.

ahundt, 2017-08-01:
The reason I was thinking about it was because reading in other languages will require much less manual code & bit fiddling for endianness if it is done with generated protobuf code. I was going to update the c++ writer/reader and implement a python reader.

If that's not worthwhile feel free to close this issue.

wohe, 2017-08-01:
I'm not sure that this is true. Especially for Python there is not much bit fiddling for endianess to do, since you have `struct.unpack`. You would just tell it to expect an 8-byte unsigned little endian integer by using the format `"<Q"`. See https://docs.python.org/3/library/struct.html

ahundt, 2017-08-07:
Cool I didn't know about that! Sounds simple enough, I'll give it a try and close this based on the discussion in my other PR https://github.com/googlecartographer/cartographer/pull/429#issuecomment-320760254.

closed 2017-08-07 by ahundt